ckb-next/ckb-next #584: [Linux] Mouse sometimes unfunctional after login

STiAT, 2020-05-13:
## General information

### Source
It's the Solus 0.4.2 package with the patches c5d3588 and 0ae46be applied (for the udev stuff we discussed 2 days ago and which was fixed by 0ae46be).
I've had this issue with vanilla 0.4.2 and a compiled master version too.

## Logs & crash reports
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/4623269/ckb-next-dev-detect-report.gz)

## What is the program's current behaviour?
Sometimes after I logon to my Desktop (KDE, Budgie, GNOME, didn't make a difference which one), I had the issue that the mouse just stops working:
- Lights disappear
- Mouse does not work

This happens in about 20 % of my reboots, so 80 % of the time it works. In any case, logging in, it seems as if the mouse gets "deactivated" and "activated" again ("default" lights go out, black for a short time, "custom" ones go on). In those 20 %, it does not recover from the "black" state, while in the other 80 % it happily loads my profile and works.

After I replug the mouse, everything is fine again, the lights go on, the mouse pointer works again.

## What is the expected behaviour of the program?
Well, that the mouse works after I logon without replugging it ;-).

## Any other notes & comments?
Sorry, none, except thanks for all your efforts!

urmamasllama, 2020-05-15:
I've had this issue with my K95W and my scimitar pro. I recently moved them both from a hub to direct ports on the motherboard. My keyboard no longer has the issue buy my mouse does. the only difference now is that my keyboard is on a usb 2.0 port and my mouse is on a usb 3.0 port. I will try moving the mouse to a usb 2.0 port soon to see if that helps. but I figured I would contribute my findings as it may help y'all track down the issue.

STiAT, 2020-05-15:
We've found out that other things possibly cause that issue. In my case it was the Citrix USB driver (Citrix Receiver), which I had installed for work purposes.

That said, they created a branch where they added some possible mutex fixes (not in master branch yet). Didn't fix it for the citrix driver, but may fix it in other circumstances.

They said they'll continue work on this since it's not that easy to counter if other drivers are interfering. I'll keep this report posted as soon as there is something new.

derbrachialist, 2020-05-29:
I am pretty sure we have the same problem: #580 can we merge the bug reports?

STiAT, 2020-06-01:
I reinstalled completely, currently, running with c5d3588 and 0ae46be against 0.4.2 i again have issues, I'll give it a shot with the patches from mutex-new-fixes again and see if we get any better results .. last time I did.

Thraix, 2020-07-05:
Might as well add that I'm having the same issue but with a corsair k70 rgb mk2. The keyboard works in the login screen (think I'm using gdm), but after login it stops responding. Disabling the ckb-next-daemon (systemctl disable ckb-next-daemon) and starting it manually after login in (systemctl start ckb-next-daemon), causes no problems for me. Not really tech-savy enough to know why this is the case, but it is a solution I've found working without having to replug the device every boot.

STiAT, 2020-07-06:
As I said, without Citrix Receiver and with the above mentioned patches this does not occur any longer.

It's likely that your distribution did not apply those patches yet, and that's why you see that behaviour (there was no new ckb-next release yet, and distros tend not to upgrade without new releases). Which distribution are you using? I should be able to check that pretty easily, and you could ask them to patch up ckb-next.

Thraix, 2020-07-09:
Sorry for the delay, apparently I don't get updates on email. I'm using arch and thought I installed it through the aur through ckb-next-git, but apparently I compiled from source myself, so I'm using an old version. Is this patch inside master yet or would I need to install from that branch?

Thraix, 2020-07-09:
Sadly the mutex-new-fixes branch did not resolve my problems. In one reboot it even disabled my keyboard before I logged in. Here is the ckb-next-dev-detect log file for reference.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/4896034/ckb-next-dev-detect-report.gz)

tatokis, 2020-07-10:
Is there a way to get either a wireshark capture or some kind of usbmon trace immediately after boot and before ckb-next-daemon starts?
I have never been able to reproduce this issue, and all I can tell is that "the device ignored/never responded to our packets".

Pillgar, 2020-07-10:
This bug sounds very similar to what I've been experiencing since I purchased the k70 mk2, though my mouse still functions {which is not a Corsair mouse), the keyboard itself does deactivate (lights turn off), then reactivates again, but 1 out of 10-20 boots/restarts, the keyboard does NOT reactivate after deactivating.  The keyboard is dead (though the numlock led is on).  This all usually happens immediately after the DE is loaded, though I remember it also happening in the grub menu the odd time, which I suspect, in my situation, could be a bug at the kernel level.  I haven't noticed any such bugs in Windows yet.

Something else to note, more often then not, when XFCE is loaded, I immediately get a pop-up that says "Sticky Keys Disabled".  It could be unrelated, but I also notice these errors in my .xsession-errors log when this happens, which all point to the keyboard and mouse.  Though they both still function when this occurs:

thunar-volman: Unsupported USB device type "usb".
thunar-volman: Unsupported input device type "/dev/input/mice".
thunar-volman: Unsupported input device type "/dev/input/event22".
thunar-volman: Unsupported input device type "/dev/input/event24".
thunar-volman: There is no device with the sysfs path "/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:1B1C:1B49.0004/input/input23".
thunar-volman: Unsupported input device type "(null)".
thunar-volman: There is no device with the sysfs path "/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:258A:0036.0003/input/input20".
thunar-volman: There is no device with the sysfs path "/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:258A:0036.0003/input/input20/event20".
thunar-volman: There is no device with the sysfs path "/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:258A:0036.0002/input/input19".
thunar-volman: There is no device with the sysfs path "/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:1B1C:1B49.0004/input/input24".
thunar-volman: Unsupported input device type "/dev/input/event21".
thunar-volman: Unsupported input device type "/dev/input/event19".
thunar-volman: Unsupported input device type "(null)".
thunar-volman: Unsupported USB device type "(null)".
thunar-volman: Unsupported USB device type "usbfs".
thunar-volman: Unsupported USB device type "usbfs".
thunar-volman: Unsupported USB device type "usb".
thunar-volman: Unsupported USB device type "usb".
thunar-volman: Unsupported USB device type "usbhid".
thunar-volman: Unsupported USB device type "usbhid".
thunar-volman: Unsupported USB device type "usb".
thunar-volman: Unsupported USB device type "hub".

I'll be upgrading my PC in a few weeks.  I'm curious to see if the problem still exists then.  There are also some reports of this happening on older Corsair keyboards in Windows.

Pillgar, 2020-07-10:
OK there definitely seems to be an issue, in my case, when both devices are plugged into the same hub, be it on the motherboard IO, USB headers, or external hub.  When both devices are using separate ports, when power and or bus is not shared, the problem doesn't occur.  This correlates with what urmamasllama mentioned also.  I narrow it down further when I find my multimeter.