ckb-next/ckb-next #586: Can't build v0.4.2

Meivyn, 2020-05-14:
I'm unable to build ``ckb-next`` package from the AUR. I also tried to build from sources at the same tagged release, I always get the same error.

```
[ 50%] Linking C executable ../../bin/ckb-next-daemon
/usr/bin/ld: CMakeFiles/ckb-next-daemon.dir/device_keyboard.c.o:(.rodata+0x0): multiple definition of `devpath'; CMakeFiles/ckb-next-daemon.dir/command.c.o:(.rodata+0x120): first defined here
/usr/bin/ld: CMakeFiles/ckb-next-daemon.dir/device_mouse.c.o:(.rodata+0x0): multiple definition of `devpath'; CMakeFiles/ckb-next-daemon.dir/command.c.o:(.rodata+0x120): first defined here
/usr/bin/ld: CMakeFiles/ckb-next-daemon.dir/devnode.c.o:(.data.rel.ro.local+0x0): multiple definition of `devpath'; CMakeFiles/ckb-next-daemon.dir/command.c.o:(.rodata+0x120): first defined here
/usr/bin/ld: CMakeFiles/ckb-next-daemon.dir/firmware.c.o:(.rodata+0x1e0): multiple definition of `devpath'; CMakeFiles/ckb-next-daemon.dir/command.c.o:(.rodata+0x120): first defined here
/usr/bin/ld: CMakeFiles/ckb-next-daemon.dir/main.c.o:(.rodata+0x8): multiple definition of `devpath'; CMakeFiles/ckb-next-daemon.dir/command.c.o:(.rodata+0x120): first defined here
/usr/bin/ld: CMakeFiles/ckb-next-daemon.dir/notify.c.o:(.rodata+0x0): multiple definition of `devpath'; CMakeFiles/ckb-next-daemon.dir/command.c.o:(.rodata+0x120): first defined here
/usr/bin/ld: CMakeFiles/ckb-next-daemon.dir/usb.c.o:(.rodata+0x40): multiple definition of `devpath'; CMakeFiles/ckb-next-daemon.dir/command.c.o:(.rodata+0x120): first defined here
/usr/bin/ld: CMakeFiles/ckb-next-daemon.dir/usb_linux.c.o:(.rodata+0xf8): multiple definition of `devpath'; CMakeFiles/ckb-next-daemon.dir/command.c.o:(.rodata+0x120): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [src/daemon/CMakeFiles/ckb-next-daemon.dir/build.make:451: bin/ckb-next-daemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:374: src/daemon/CMakeFiles/ckb-next-daemon.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 50%] Built target ckb-next_autogen
make: *** [Makefile:172: all] Error 2
An error occurred,
```

Building ``master`` works, but I'd like to stay on a stable release.

ZirconiumX, 2020-05-14:
Actually master literally just merged in the fix for it; you can apply the
fix if you really want to use 0.4.2.

However, ckb-next doesn't have "stable releases", just "releases". The only
thing we ever really support is master.

ZirconiumX, 2020-05-14:
Either way, the fix for this is to use master.

closed 2020-05-14 by ZirconiumX

HeroCC, 2020-05-26:
I'm affected by this as well, and it looks like it has been a while since there has been an official release, with multiple new features since then. Is it possible to tag a new release? 0.4.2 or 0.5.0 perhaps?