ckb-next/ckb-next #589: Support Corsair Vengeance RGB Pro RAM?

pa3ng, 2020-05-19:
# Before you proceed

- Check if your device is listed in the [wiki](https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware).
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

---

Enter the name of your device here and attach the report generated above.

owendelong, 2020-05-29:
Likely the same (or very similar) code could support the Dominator series as well. Please consider validating against that hardware as well.

Corsair also makes a series of fans and other chassis accoutrement with controllable RGB LEDs.
It would be nice to see ckb-next eventually encompass all that can be controlled by iCUE.

gravityfargo, 2023-04-21:
This duplicate to #316  but keeping it open since it's a request for a specific model.

just-Smidge, 2025-01-01:
is there any other way to controll vengeance ram on linux?