ckb-next/ckb-next #590: Lighting Node PRO

Shaneee, 2020-05-19:
Would it be possible to get the Lighting Node PRO supported? I've added my old RGB strips to my Plex computer under my desk and would like to control them. I've attached the report.
Thanks
[ckb-next-report-1589854363.gz](https://github.com/ckb-next/ckb-next/files/4647890/ckb-next-report-1589854363.gz)

thespyboss, 2020-05-19:
@Shaneee i think it is supported but you need to look through the #453 and i think you can get it to work. I have lighting node pro but hooked up to one LL120 fan. If the rgb strips work the same as the fan leds, then your set.

Shaneee, 2020-05-19:
> @Shaneee i think it is supported but you need to look through the #453 and i think you can get it to work. I have lighting node pro but hooked up to one LL120 fan. If the rgb strips work the same as the fan leds, then your set. All you need is to install that ckb version that has the node pro support in it. Just ping me if you get any problems.

The problem is I'm unable to compile the source on Ubuntu. Here are the logs.
[CMakeError.log](https://github.com/ckb-next/ckb-next/files/4650497/CMakeError.log)
[CMakeOutput.log](https://github.com/ckb-next/ckb-next/files/4650498/CMakeOutput.log)

thespyboss, 2020-05-19:
i cant really tell whats going on in those files, but are you just trying to install using the .zip file? like how you would normally install ckb?

thespyboss, 2020-05-19:
i was going to say, if you can hop on the irc chat i can help out a little quicker than reloading the webpage

Shaneee, 2020-05-19:
Had apt issues. Fixed with aptitude and added modifications to source from mrdev023. All working now.

closed 2020-05-19 by Shaneee

etrigan63, 2020-05-19:
Pray elucidate as CKB-Next on my system still fails to recognize the
Lighting Node Pro on my system even though it is listed in lsusb.

Carlos Echenique Photographer, CE Photographic

305.219.2433 | carlos@echenique.com
Website: https://www.cephotographic.com
<https://www.facebook.com/echeniquephotos>
<https://flickr.com/photos/echenique> <https://instagram.com/cephotographic>