ckb-next/ckb-next #591: ./quickinstall not working

itzvoxxie, 2020-05-21:
When I first try to use quick isntall, it says _permission denied_ so from there I use _cmod +x_ and then after that I type in _ls_ which shows me the files. I try to use _./quickinstall_
and I get this ---
_./quickinstall: line 51: cmake: command not found
An error occurred,
press enter to exit._
Im kinda new to this so I am very confused!

ZirconiumX, 2020-05-21:
You need to install cmake from your package manager. What distro are you on?

itzvoxxie, 2020-05-21:
> You need to install cmake from your package manager. What distro are you on?

idk what a distro is lmao, but i got passed it, i fiddle around with some old tricks and got through, but now im hitting this error-----------
_make[2]: Leaving directory '/home/snipefractons/ckb-next/src/ckb'
make[1]: Leaving directory '/home/snipefractons/ckb-next/src/ckb'
kbbindwidget.ui: Warning: Z-order assignment: 'verticalSpacer' is not a valid widget.
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [Makefile.Release:1144: release/kbwidget.o] Error 1
make[2]: *** Waiting for unfinished jobs....
virtual memory exhausted: Cannot allocate memory
make[2]: *** [Makefile.Release:1226: release/animscript.o] Error 1
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
make[2]: *** [Makefile.Release:1315: release/kbprofiledialog.o] Error 1
make[2]: *** [Makefile.Release:1178: release/kblightwidget.o] Error 1
virtual memory exhausted: Cannot allocate memory
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [Makefile.Release:1122: release/mainwindow.o] Error 1
make[2]: *** [Makefile.Release:1233: release/kbanim.o] Error 1
make[1]: *** [Makefile:40: release] Error 2
make: *** [Makefile:77: sub-src-ckb-make_first] Error 2
_

any ideas?

ZirconiumX, 2020-05-21:
A distro is a Linux distribution, unless you're on Mac.

The errors you're having *there* are that you're running out of memory.

itzvoxxie, 2020-05-21:
I got this chromebook for free, freshly bought and they had no use, and im running on the Linux beta lmao. Im sold my pc, because im getting a new one built,
and how much memory does it need do you know? bc i should i have plenty

and thanks for the quick response

tatokis, 2020-05-21:
You can open up quickinstall in a text editor and replace `-j "$JOBS"` with `-j 1`.
It will compile much slower but use less memory.

itzvoxxie, 2020-05-21:
> You can open up quickinstall in a text editor and replace `-j "$JOBS"` with `-j 1`.
> It will compile much slower but use less memory.

so just remove the "$JOBS" part?

ZirconiumX, 2020-05-21:
No, because that's `-j` (which uses all threads at once and drains your
memory) not `-j 1` (which uses a single thread).

itzvoxxie, 2020-05-21:
oh okay, thank you, and sorry for the bombard of questions, im just now getting into all this lol

itzvoxxie, 2020-05-21:
> You can open up quickinstall in a text editor and replace `-j "$JOBS"` with `-j 1`.
> It will compile much slower but use less memory.

i might seem slow saying this but, i cant seem to find that code in `quickinstall`

itzvoxxie, 2020-05-21:
> You can open up quickinstall in a text editor and replace `-j "$JOBS"` with `-j 1`.
> It will compile much slower but use less memory.

i am now getting this error

```
  Failed to find Qt5 component "X11Extras" config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt5X11Extras/Qt5X11ExtrasConfig.cmake"

  

-- Checking for module 'xcb'
--   Found xcb, version 1.13.1
-- Checking for module 'xcb-ewmh'
--   No package 'xcb-ewmh' found
-- Checking for module 'xcb-screensaver'
--   No package 'xcb-screensaver' found
CMake Error at src/gui/CMakeLists.txt:255 (message):
  XCB-Screensaver or Qt5 X11Extras was not found.  Either install them or
  pass -DUSE_XCB=0 to disable idle detection.

```

ZirconiumX, 2020-05-21:
Well, you don't have *all* of Qt5 installed, it seems.

itzvoxxie, 2020-05-21:
> Well, you don't have _all_ of Qt5 installed, it seems.

how do i get them all lol

itzvoxxie, 2020-05-21:
> Well, you don't have _all_ of Qt5 installed, it seems.

Im failing to find the following--------- `Qt5X11Extras_DIR`

closed 2020-05-21 by itzvoxxie