ckb-next/ckb-next #593: Not Showing Up

itzvoxxie, 2020-05-21:
It says it is installed, but I cant find it anywhere?

ZirconiumX, 2020-05-21:
Run `ckb-next` from the command line.

itzvoxxie, 2020-05-21:
Its saying permission denied, should I just use `chmod +x ./ckb-next`??

ZirconiumX, 2020-05-21:
It should have set it up, although I'm not sure how to autostart things on
Chrome OS. If you get "permission denied" that means it somehow failed to
build?

itzvoxxie, 2020-05-21:
im goinng to try it bc i had to get through quickinstall using chmod +x

itzvoxxie, 2020-05-21:
```
QObject::startTimer: Timers cannot have negative intervals
Path to settings: "/home/snipefractons/.config/ckb-next/ckb-next.conf"
Looking for old settings in: "/home/snipefractons/.config/ckb/ckb.conf"
Using AppIndicator
Scanning "/home/snipefractons/ckb-next/CMakeLists.txt"
"" has no default preset and will not be loaded.
Scanning "/home/snipefractons/ckb-next/quickinstall"
QProcess: Destroyed while process ("/home/snipefractons/ckb-next/quickinstall") is still running.
"" has no default preset and will not be loaded.
Scanning "/usr/local/libexec/ckb-next-animations/gradient"
Scanning "/usr/local/libexec/ckb-next-animations/heat"
Scanning "/usr/local/libexec/ckb-next-animations/invaders"
Scanning "/usr/local/libexec/ckb-next-animations/life"
Scanning "/usr/local/libexec/ckb-next-animations/mviz"
Scanning "/usr/local/libexec/ckb-next-animations/pinwheel"
Scanning "/usr/local/libexec/ckb-next-animations/pipe"
Scanning "/usr/local/libexec/ckb-next-animations/rain"
Scanning "/usr/local/libexec/ckb-next-animations/random"
Scanning "/usr/local/libexec/ckb-next-animations/ripple"
Scanning "/usr/local/libexec/ckb-next-animations/snake"
Scanning "/usr/local/libexec/ckb-next-animations/wave"
Downloaded new firmware list. 38 entries found.
```
is what im got after and then it loaded up saying i dont have dameon

ZirconiumX, 2020-05-21:
Then in another terminal, run `sudo ckb-next-daemon`. You might have to
`chmod +x` it. This is a very temporary solution though; you need to get
`ckb-next-daemon` running on boot, and `ckb-next` running on desktop start.

itzvoxxie, 2020-05-21:
I got this--------- sudo ckb-next-daemon
ckb-next: Corsair RGB driver 0.4.2-unknown
[E] _mkdevpath (devnode.c:204): Unable to create /dev/input/ckb0: No such file or directory
-----------------
and how do get the on start?

closed 2020-05-21 by itzvoxxie