ckb-next/ckb-next #594: Im need help, im confused.

itzvoxxie, 2020-05-21:
Uhh??--------------------------------------------------->
`Done! Please send the following report to the developers.
Location: /home/snipefractons/ckb-next/ckb-next-dev-detect-report.gz
./ckb-next-dev-detect-report: line 128: Kernel:: command not found
./ckb-next-dev-detect-report: line 130: OS: command not found
./ckb-next-dev-detect-report: line 133: USB: command not found
./ckb-next-dev-detect-report: line 136: Systemctl: command not found
./ckb-next-dev-detect-report: line 139: Dev: command not found
./ckb-next-dev-detect-report: line 140: find:: command not found
./ckb-next-dev-detect-report: line 142: Environment:: command not found
./ckb-next-dev-detect-report: line 151: Kernel: command not found
./ckb-next-dev-detect-report: line 152: snd_intel8x0.inside_vm=1: command not found
./ckb-next-dev-detect-report: line 154: GUI: command not found
./ckb-next-dev-detect-report: command substitution: line 155: syntax error near unexpected token `0x00007ffedf951000'
./ckb-next-dev-detect-report: command substitution: line 155: ` linux-vdso.so.1 (0x00007ffedf951000)'`

ZirconiumX, 2020-05-21:
You seem to be trying to execute a text file as a shell script. It's not a shell script, but just plain text.

itzvoxxie, 2020-05-21:
> You seem to be trying to execute a text file as a shell script. It's not a shell script, but just plain text.

What does that mean? What do I do?

closed 2020-05-21 by itzvoxxie