ckb-next/ckb-next #595: Cannot Change Key LED RGB Values on First-gen Corsair K95 RGB

InklingGirl, 2020-05-25:
The programme gives me an error prompt & then hangs before crashing.

Output:

fg@fg-Z87X-UD5-TH:~$ ckb-next
QObject::startTimer: Timers cannot have negative intervals
Gtk-Message: 05:19:45.894: Failed to load module "canberra-gtk-module"
Path to settings: "/home/fg/.config/ckb-next/ckb-next.conf"
Using QSytemTrayIcon
Scanning "/usr/lib/ckb-next-animations/gradient"
Scanning "/usr/lib/ckb-next-animations/heat"
Scanning "/usr/lib/ckb-next-animations/invaders"
Scanning "/usr/lib/ckb-next-animations/life"
Scanning "/usr/lib/ckb-next-animations/mviz"
Scanning "/usr/lib/ckb-next-animations/pinwheel"
Scanning "/usr/lib/ckb-next-animations/pipe"
Scanning "/usr/lib/ckb-next-animations/rain"
Scanning "/usr/lib/ckb-next-animations/random"
Scanning "/usr/lib/ckb-next-animations/ripple"
Scanning "/usr/lib/ckb-next-animations/snake"
Scanning "/usr/lib/ckb-next-animations/wave"
Downloaded new firmware list. 38 entries found.

(ckb-next:27812): GLib-GObject-WARNING **: 05:20:06.741: cannot register existing type 'GtkWidget'

(ckb-next:27812): GLib-GObject-WARNING **: 05:20:06.741: cannot add class private field to invalid type '<invalid>'

(ckb-next:27812): GLib-GObject-WARNING **: 05:20:06.741: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(ckb-next:27812): GLib-GObject-CRITICAL **: 05:20:06.741: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb-next:27812): GLib-GObject-WARNING **: 05:20:06.741: cannot register existing type 'GtkBuildable'

(ckb-next:27812): GLib-GObject-CRITICAL **: 05:20:06.741: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed

(ckb-next:27812): GLib-CRITICAL **: 05:20:06.741: g_once_init_leave: assertion 'result != 0' failed

(ckb-next:27812): GLib-GObject-CRITICAL **: 05:20:06.741: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb-next:27812): GLib-GObject-CRITICAL **: 05:20:06.741: g_type_register_static: assertion 'parent_type > 0' failed

(ckb-next:27812): GLib-GObject-WARNING **: 05:20:06.741: cannot add private field to invalid (non-instantiatable) type '<invalid>'

ZirconiumX, 2020-05-25:
This is a common enough issue that we put in that error prompt especially for it. Please read the section there first.

InklingGirl, 2020-05-25:
> This is a common enough issue that we put in that error prompt especially for it. Please read the section there first.

OK, done! Now please help me. 'Common' enough it probably shouldn't be present to begin with, especially with how far back some of these issues are dated. I've looked over the issue discussions & documentation, I've installed qt5ct & other such things but it's still broken. 

I'm a beginner user, so there were some things I couldn't wrap my head around, so I may very well be missing something, so please assist me in fixing the issue. Thank you!

tatokis, 2020-05-25:
First of all, generate and attach a report, as you were asked by the issue template.
Second, assuming you're on ubuntu, have you logged out and back in after running the commands?

And finally, `'Common' enough it probably shouldn't be present to begin with, especially with how far back some of these issue reports are dated.` if it was in our control, it wouldn't happen.
Feel free to complain to your distro maintainers for not backporting the fixes.

ZirconiumX, 2020-05-25:
> 'Common' enough it probably shouldn't be present to begin with, especially with how far back some of these issue reports are dated.

There isn't anything ckb-next can do to fix the problem: your system is misconfigured, and that's why the wiki tells you to:
- download `qt5ct` (which you say you've done)
- add the string `export QT_QPA_PLATFORMTHEME=qt5ct` to your `~/.profile` so it gets ran when you log in
- log out and back in again
- run `qt5ct` from a terminal
- select `gtk2` from the Style dropdown menu and click ok
- re-run ckb-next

InklingGirl, 2020-05-25:
[ckb-next-report-1590435431.gz](https://github.com/ckb-next/ckb-next/files/4678636/ckb-next-report-1590435431.gz)

Yes logging in & out seemed to work.

[image]

This can be closed.

closed 2020-05-25 by ZirconiumX

ZirconiumX, 2020-05-25:
I have reworded the [wiki page](https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#glib-critical-errors-and-lock-up-at-colour-chooser) to be clearer.

tatokis, 2020-05-25:
I will also have to take back what I said, because apparently while the issue was never fixed, it does occur because we link against appindicator.

It is not the distro's fault after all.