ckb-next/ckb-next #597: Report all missing dependencies (ideally by package name) before exiting

owendelong, 2020-05-28:
## What is the program's current behaviour?
quickinstall reports (somewhat ambiguously) the first missing dependency (group) it encounters and then exits. It does not report dependencies by installable package names, leaving the user to hunt for package sets that might provide the required functionality and in some cases repeated guess work is required until finding the right one(s).

## What would you like it to do instead?
Ideally, detect and report all of the missing dependencies prior to exiting. In a truly marvelous world, this would include reporting the packages that would need to be installed to satisfy the dependencies and perhaps even the option to install them automatically prior to continuing.

## Any other notes & comments?
Thanks!

tatokis, 2020-05-28:
This isn't really possible. Each distro has its own package names, and some packages are split up into more packages on other distros.

Quickinstall itself doesn't report anything, it just calls cmake. All it can tell is if something went wrong or not.

Even if we were to add custom error messages in cmake that included all possible package names for popular distros (which would be a bit of a maintenance nightmare), we still can't check if they are all installed at once.

That would require detecting what the default package manager is, what the packages have to be for that distro, for that specific version of the distro, and then query the package manager.

This is why this page was created https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#build-from-source

So while I'm not willing to implement anything like this programmatically, I'll leave this issue open as I'm willing to make quickinstall print a message containing that link and a generic message telling people to ensure they have all the dependencies installed as per that link.

owendelong, 2020-05-28:
Fair enough...

A message saying "Refer to <link> for information on dependencies for popular distros" would have been far more useful than the message currently printed.

I will say that if the format of that page remained consistent, it would be possible to script the translation of that page into a query for each of the distros covered, so the maintenance "nightmare" seems on par with maintaining the wiki page.