ckb-next/ckb-next #606: Option to start minimized in system tray

mightymouth, 2020-06-14:
Rather than have the application open a window have the option to open minimized in the system tray.

ZirconiumX, 2020-06-14:
That would be `ckb-next -b`, right?

mightymouth, 2020-06-14:
> That would be `ckb-next -b`, right?

Yes, but have a tick box option in the GUI.

jannis3005, 2020-07-27:
> That would be `ckb-next -b`, right?

But this does already work, i think. It's just the GUI Switch that's missing

[image]

mightymouth, 2020-07-27:
The option is there but it does nothing. it still always opens full screen at login.

jannis3005, 2020-07-27:
> The option is there but it does nothing. it still always opens full screen at login.

What Distro + DE are you using? For me (Arch + KDE) it works perfectly and just starts in background

mightymouth, 2020-07-27:
Kubuntu 20.04. I have tried on KDE Neon also which was the same which makes sense as it is essentially the same.

tatokis, 2020-07-27:
KDE is ignoring the autostart entry and trying to restore the session. In doing so, it passes its own arguments that ckb-next ignores due to the way it's set up. There's a fix for this in master.
https://github.com/ckb-next/ckb-next/issues/510

closed 2020-07-27 by tatokis

mightymouth, 2020-07-27:
I have installed master and it is indeed working as it should. thank you.