ckb-next/ckb-next #608: Ironclaw RGB: DPI RGB indicators one stage ahead

mlugg, 2020-06-22:
Hi,

Using the Ironclaw RGB, I've noticed an issue where the RGB DPI indicators are off by one stage compared to the actual DPI. For instance, I have 3 DPI settings, where the lowest is red, the middle is yellow, and the highest is green; when I'm on the first DPI setting, only 1 indicator is on as expected, but is it yellow. Similarly, on the second setting, two lights are on as expected, but they appear green; and in the third and final setting, all three lights are on, but they appear red.

I've also had these indicators entirely turn off at seemingly random times, which seems to fix itself if you toggle the setting a few times.

I got ckb-next from the Void Linux repos, but I've confirmed the issue is still present in master.

[Generated report](https://mlugg.co.uk/ckb-next-report-1592823187.gz)

Thanks

Ravenslofty, 2020-07-19:
Does the Ironclaw only offer three DPI modes (it only has three LEDs)? All others we've had are five, so the Ironclaw would need a workaround in that case.

mlugg, 2020-07-20:
It only has 3 LEDs, but it's capable of 5 modes (the fourth and fifth have all the LEDs on, like in the third state). I normally only have 3 modes enabled, but if I turn on the others, the colors for those two are actually correct - it's only modes 1-3 which have incorrect LED colors.

mlugg, 2020-12-31:
A small thing I've recently noticed: if the lights turn off due to being idle, when they turn back on, the indicator LEDs are the correct color (until they're changed again). Not sure if this helps with debugging at all? (I can provide any more reports or info if neccesary)

mlugg, 2021-02-14:
Slight update to my last comment: I can also actually make it happen by manually toggling the brightness through 0%. Once the lights turn back on, they're correct until the DPI changes.