ckb-next/ckb-next #609: Installation fails due to missing 'Release file'

PoLabs, 2020-06-23:
Attempting to install via latest release using:

sudo add-apt-repository ppa:tatokis/ckb-next

Where I encounter this error: 

E: The repository 'http://archive.ubuntu.com/ubuntu oneiric Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Also tried the git-master version and building from source but they all experienced errors in a similar vein.

tatokis, 2020-06-28:
Ubuntu 11.10 reached end of life almost 7 years ago. Upgrade ASAP.
We do not, and will not support old versions.

closed 2020-06-28 by tatokis

PoLabs, 2020-06-28:
I'm on Ubuntu 18

Ravenslofty, 2020-06-28:
`oneiric` is the codename for Ubuntu 11.10 Oneiric Ocelot. Either you had that and never updated, or you've somehow broken your APT installation.

tatokis, 2020-06-28:
```
sudo add-apt-repository ppa:tatokis/ckb-next-git
sudo apt-get update
```

Remove the existing PPA and run the above