ckb-next/ckb-next #611: redhat/centos can't compile projet because of _Atomic

Isnake, 2020-06-24:
Hi,

when i try to do the quickinstall on redhat 7 i have the error 

ckb-next/src/daemon/main.c:9:16: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘int’
 extern _Atomic int reset_stop;
                ^
/home/samie.saouchi/Documents/corsair2/ckb-next/src/daemon/main.c: In function ‘quit’:
/home/samie.saouchi/Documents/corsair2/ckb-next/src/daemon/main.c:30:5: error: ‘reset_stop’ undeclared (first use in this function)
     reset_stop = 1;

Its the same everytime with the _Atomic.

Ravenslofty, 2020-06-24:
Your compiler doesn't support C11. You need to upgrade it to build ckb-next.

Isnake, 2020-06-25:
Thanks ! i'll try to update that

Isnake, 2020-06-29:
I've resolved it by : 
yum install (all dependency of ckb-next)
yum install devtoolset-9
scl enable devtoolset-9 'bash'
./quickinstall

closed 2020-06-29 by Isnake