ckb-next/ckb-next #613: Where are profiles stored and how do I copy them?

CrushedPixl, 2020-06-26:
I didn't export my profiles before reinstalling my OS. I have a full system backup, but I don't know which file to copy. I've tried copying `~/.config/ckb-next/ckb-next.conf` and restarting with `ckb-next -c && rc-service ckb-next-daemon restart && ckb-next`, but nothing seems to have changed.

Additionally, I tried to trace which files were accessed through `inotifywait -m -r -e OPEN / | grep ckb` without success. The only accessed files other than the config mentioned above were in `/dev/input/ckb0/` and  `/usr/lib/ckb-next-animations/` when switching profiles.

tatokis, 2020-06-27:
It is `~/.config/ckb-next/`.
The problem is that you are replacing it with ckb-next (the GUI) running.

Close it, then replace it, then restart it.

CrushedPixl, 2020-06-27:
This worked. Thanks!

closed 2020-06-27 by CrushedPixl