ckb-next/ckb-next #614: Ubuntu 20.04 - K95 - No Devices Connected

teknofile, 2020-06-27:
### Source
How and where did you get this program, e.g.:
Tried both the PPA for Ubuntu and also building from source.

## Logs & crash reports

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behaviour?
New K95 Platinum Gaming Keyboard. "No devices connected" when runnign ckb-next.

## What is the expected behaviour of the program?
It detects and lets me fidle with the keyboard.

## Any other notes & comments?
The keyboard still works as a HID device. 

[ckb-next-report-1593285313.gz](https://github.com/ckb-next/ckb-next/files/4841113/ckb-next-report-1593285313.gz)

[image]

tatokis, 2020-06-27:
You have a K95 Platinum XT which is not supported

teknofile, 2020-06-29:
Ah, thanks. I didn't realize that there was a difference between the "Platinum" and "Platinum XT". Started looking at the code briefly this weekend and adding a new device ID didn't look too complicated. If it's just a matter of that, I could probably get that done and issue a PR.

tatokis, 2020-06-29:
The XT uses the new protocol that we have not implemented.
It is not that simple sadly

pinheadmz, 2020-06-30:
I also have a `Corsair Gaming K95 RGB PLATINUM Keyboard` on a Lenovo Y740 laptop. Tracking this issue! Might look into contributing if I can

teknofile, 2020-07-01:
> The XT uses the new protocol that we have not implemented. There's some very slow work in progress.
> It is not that simple sadly

Well, that's stinky. 

Has any documentation already began about the new protocol?

Ravenslofty, 2020-07-01:
We don't know for sure, but we expect significant overlaps with the
Ironclaw Wireless, as both identify as the same protocol version.

ghost1347, 2020-07-11:
Damn I got an xt not knowing that made a difference

tatokis, 2020-10-19:
https://github.com/ckb-next/ckb-next/issues/614

closed 2020-10-19 by tatokis