ckb-next/ckb-next #617: K95 Keyboard unresponsive when launching ckb-next

ArcticStag, 2020-07-02:
`                  /:-------------:\          
               :-------------------::                       ---------------------------------- 
     :-----------/shhOHbmp---:\                    OS: Fedora 32 (KDE Plasma) x86_64 
   /-----------omMMMNNNMMD  ---:          Kernel: 5.7.6-201.fc32.x86_64 
  :-----------sMMMMNMNMP.    ---:           Uptime: 24 mins 
 :-----------:MMMdP-------    ---\                 Packages: 1825 (rpm) 
,------------:MMMd--------    ---:                  Shell: bash 5.0.17 
:------------:MMMd-------    .---:                  Resolution: 1920x1080, 1920x1080 
:----    oNMMMMMMMMMNho     .----:   DE: Plasma 
:--     .+shhhMMMmhhy++   .------/        WM: KWin 
:-    -------:MMMd--------------:                 WM Theme: Sweet-Dark 
:-   --------/MMMd-------------;                 Theme: Sweet [Plasma], Adwaita [GTK2] 
:-    ------/hMMMy------------:                  Icons: candy-icons [Plasma], candy-icons [GTK2/3] 
:-- :dMNdhhdNMMNo------------;          Terminal: terminator 
:---:sdNMMMMNds:------------:             CPU: AMD Ryzen 7 1800X (16) @ 3.600GHz 
:------:://:-------------::                             GPU: NVIDIA GeForce GTX 1080 Ti 
:---------------------://                               Memory: 3071MiB / 32097MiB 
`

When I launch the KCB-Next application the keyboard no longer functions. Restarting the machine is the only way to regain functionality

derbrachialist, 2020-07-08:
Maybe the same problem I have in #580? I still have no solution other than unplugging and plugging it back in, have you tried that?

kxr, 2020-12-24:
Facing the same problem. The moment I launch kcb-next app, the keyboard stops working. I have to unplug and re-plug to get it to start working.

OS: Fedora 32 (KDE Plasma) x86_64
Kernel: 5.9.13-100.fc32.x86_64
CPU: Intel(R) Core(TM) i7-6820HQ
Memory: 64 GB
PC: Lenovo P50 Workstation (laptop)

Package: ckb-next-0.4.2-3.fc32 (from @fedora repository)