ckb-next/ckb-next #618: Long shutdown/reboot time

zany130, 2020-07-11:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**
this did not work for me it froze when it launched the file browser 

## General information
same as issue #168
### Source
How and where did you get this program, e.g.:
  * Package from AUR/PPA/COPR/...
AUR ckb-next-git
  * Built manually using quickinstall script
  * Pre-built dmg for macOS

## Logs & crash reports

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behaviour?
takes about 4 minutes to shut down
## What is the expected behaviour of the program?
takes less than a minute to shutdown
## Any other notes & comments?

zany130, 2020-08-07:
can confirm that it seems ckb-next as well as other processes are at fault

tatokis, 2020-08-07:
There's nothing we can do unfortunately, other than maybe add an option to just not reattach the device back to the kernel.
There's a macro in usb.h you can add your device to, and that will "fix" it.

https://github.com/ckb-next/ckb-next/blob/master/src/daemon/usb.h#L192