ckb-next/ckb-next #619: Testing and Validation

neeldug, 2020-07-17:
## What is the program's current behaviour?
Currently there is no automated testing and validation, which was previously mentioned in #12 however this was closed in the end as Travis CI was unable at the time to test on various versions of macOS and XCode, this has since changed and so I was thinking that perhaps we should consider resurrecting this issue.
## What would you like it to do instead?
Setup a basic Travis CI workflow for PRs and Pushes that runs it against as many versions as the core maintainers would like. The implementation for using many different versions of macOS and XCode is documented very well [here](https://docs.travis-ci.com/user/multi-os/#example-multi-os-build-matrix), currently homebrew makes use of this multi os build matrix to test new packages.
## Any other notes & comments?

Ravenslofty, 2020-07-19:
So, I've been thinking about this.

Travis is not very helpful in this situation; whether the code compiles has no correlation to whether it works and the only way we can check if it works is by testing against real hardware.

neeldug, 2020-07-19:
While this is true, it would still allow for basic testing to be done prior to merging any changes, and it allows testing for compilation on various platforms, it would be impossible to setup CI to test with the hardware, but at least this ensures some level of compatibility.