ckb-next/ckb-next #622: AmazonBasics mechanical keyboard support

koroki, 2020-07-21:
Amazon is selling with its our brand a mechanical keyboard that is a clone of Corsair K95 RGB Platinum:

https://www.amazon.es/dp/B07D91YT8G
https://www.amazon.de/dp/B07D93KC21/
https://www.amazon.co.uk/dp/B07D8X6CK8

I just bought and the USB ID is:  3938:1150. There is not text description in lsusb.

Ravenslofty, 2020-07-21:
I'm a bit skeptical it will speak the same protocol as Corsair as we don't know the internals.

Can you run `lsusb -vd 3938:1150`?

koroki, 2020-07-21:
here:
Bus 002 Device 007: ID 3938:1150  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x3938 
  idProduct          0x1150 
  bcdDevice            0.08
  iManufacturer           1 Acrox
  iProduct                2 AmazonBasics gaming keyboard
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     471
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

Ravenslofty, 2020-07-21:
Sorry to disappoint you, but this very definitely isn't a K95 RGB Platinum clone:

- Corsair keyboards have multiple endpoints for CUE/ckb-next to transfer RGB data to, and this has only one.
- The firmware version (`bcdDevice`) number (0.08) is nothing like Corsair's versioning system.
- The `iManufacturer` is `Acrox`, not `Corsair`. 
- There's no serial number string.

You can see what the expected output for a genuine K95 RGB Platinum is [here](https://github.com/ckb-next/corsair-protocol/blob/master/lsusb/1b1c/1b2d/0107).

With this in mind, this keyboard would probably require reverse engineering through Wireshark traces, and what little energy we presently have for that is being spent on Corsair's Bragi protocol.

closed 2020-07-21 by Ravenslofty

koroki, 2020-07-21:
Ok! thanks to check it.