ckb-next/ckb-next #623: Windows port ?

Gomam0n-afk, 2020-07-21:
## What is the program's current behaviour?
No windows version
## What would you like it to do instead?
have a windows version
## Any other notes & comments?
Corsair's official iCUE software is just an abomination, it's currently sitting at 130mb of ram taken just to make my LEDs light up ! i found ckb-next to be insanely superior to Corsair's official program, so i wanted to know if it was possible to have a windows version of ckb-next somewhere down the line ? and is there any alternative i can use while waiting for that ?

pic attached : ram consumption by ckb-next as of now;

I would also like to thank you for your work so far, ckb-next never disapointed me, and has been a go to software for everytime i install any linux based OS.
[image]

Ravenslofty, 2020-07-21:
We do actually have an experimental Windows port in the windows branch, but it requires downloading a bunch of external software to use other people's drivers because we can't afford a code signing certificate for our own USB kernel driver.

@tatokis wrote it and would probably know more.

cela96, 2020-07-29:
I have Corsair K95 RGB buyed February 2016 and on windows i'm still obliged to use the old CUE because the new iCUE is buggy, keep crashing and not work with some marco that i use. It would be a nice thing to have an alternative open source app. 

Stupid question there is no way to use corsair driver and integrate them in cbk-next?

dschoerk, 2020-09-03:
i've recently discovered ckb-next for linux, since i run windows as well for some games i would highly appreciate a windows port! iCUE sadly is very slow and takes a lot of resources.

trajano, 2021-05-19:
@cela96 iCUE 4 seems to be a bit more stable for me.

cela96, 2021-05-19:
@trajano thanks, i will definitely try it

maffe, 2022-07-21:
Duplicate of https://github.com/ckb-next/ckb-next/issues/301

closed 2022-07-23 by tatokis