ckb-next/ckb-next #624: support request Scimitar RGB Elite (2020)

middleoutworks, 2020-07-25:
I purchased Corsair Scimitar RGB Elite (new 2020 version) hoping it would just work with the app; it isn't recognized and I've tried the trouble shooting steps. Installed from Linux Mint 20 Software manager, Settings tab shows up but "No device connected" doesn't go away after multiple USB port swaps, reinstalls, & reboots. I'm hoping there is some code I can add to put this device in the registry list of app. Any help would be greatly appreciated.

Ravenslofty, 2020-07-25:
Try rebuilding ckb-next from latest git master; we added it there about a
week ago.

middleoutworks, 2020-07-25:
I've no idea how to do any of this git master stuff. So I uninstalled the app and then I did this: `sudo add-apt-repository ppa:tatokis/ckb-next-git
sudo apt-get update`and then this `sudo apt-get install build-essential cmake libudev-dev qt5-default zlib1g-dev libappindicator-dev libpulse-dev libquazip5-dev` so I needed to install git `sudo apt install git` and then I cloned `git clone https://github.com/ckb-next/ckb-next.git` CD'd and did `./quickinstall` and I got these errors 

> CMake Warning at src/gui/CMakeLists.txt:102 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake

  but it set Qt5_FOUND to FALSE so package "Qt5" is considered to be NOT
  FOUND.  Reason given by package:

  Failed to find Qt5 component "X11Extras" config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt5X11Extras/Qt5X11ExtrasConfig.cmake"

  

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'xcb'
--   Found xcb, version 1.14
-- Checking for module 'xcb-ewmh'
--   No package 'xcb-ewmh' found
-- Checking for module 'xcb-screensaver'
--   No package 'xcb-screensaver' found
CMake Error at src/gui/CMakeLists.txt:255 (message):
  XCB-Screensaver or Qt5 X11Extras was not found.  Either install them or
  pass -DUSE_XCB=0 to disable idle detection.

-- Configuring incomplete, errors occurred!
See also "/home/cinnamon/ckb-next/build/CMakeFiles/CMakeOutput.log".
See also "/home/cinnamon/ckb-next/build/CMakeFiles/CMakeError.log".
An error occurred,
press enter to exit.

Yeah brand new linux user here, sorry but I don't understand this. Any way I can just install something by downloading a file?

Ravenslofty, 2020-07-25:
Try `sudo systemctl unmask ckb-next-daemon.service`.

Ravenslofty, 2020-07-25:
Although you might want to wait a bit; there's a new build currently building.

Ravenslofty, 2020-07-25:
Okay, so, you can grab a `.deb` file from [here](https://launchpad.net/~tatokis/+archive/ubuntu/ckb-next-git/+build/19638959/+files/ckb-next_0.4.2-63-gce3747d~focal_amd64.deb) and install it with `sudo dpkg -i ckb-next_0.4.2-63-gce3747d~focal_amd64.deb`.

middleoutworks, 2020-07-25:
Thank you !!! the deb worked. the PPA install just gave me version 58, but this v63 works. 58 still didn't support my mouse. Resolved !

Ravenslofty, 2020-07-25:
Let us know if you see anything odd; the Scimitar Elite support is beta-quality at the moment.

closed 2020-07-25 by Ravenslofty