ckb-next/ckb-next #629: IronClaw via USB

thomas-wach, 2020-08-06:
Hi,
 in usb.h I found:

#define P_IRONCLAW           0x1b5d
#define IS_IRONCLAW(kb)      ((kb)->vendor == V_CORSAIR && (kb)->product == P_IRONCLAW)

In my log I see:

usb 8-1.2.3: new full-speed USB device number 37 using ehci-pci
usb 8-1.2.3: New USB device found, idVendor=1b1c, **idProduct=1b4c**, bcdDevice= 1.16
usb 8-1.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 8-1.2.3: Product: CORSAIR IRONCLAW RGB WIRELESS Gaming Mouse
usb 8-1.2.3: Manufacturer: Corsair

Can direct USB-connection simply be covered by adding "1b4c" to the list of devices?

Ravenslofty, 2020-08-06:
The Ironclaw Wireless speaks an entirely different protocol to the Ironclaw (Wired), and there's a bounty (#558) on it; if it was this easy to add support for the Ironclaw Wireless we'd have already done it by now.

closed 2020-08-06 by Ravenslofty

thomas-wach, 2020-08-07:
Ah, sry.
I thought "IronClaw Wireless" with switch set to USB-only is identical to a "plain" IronClaw-USB-only.