ckb-next/ckb-next #630: Ckb-next icon in the tray has a breakdown in Ubuntu Budgie

projectintel-anon, 2020-08-06:
Loading Ckb-next in Ubuntu Budgie causes the app icon in the tray to take up half your screen... and I can't seem to turn it off apart from closing the app and not using it.

Please help!

[image]

Ravenslofty, 2020-08-06:
I'll reply to this in a bit, but holy shit that's hilarious.

projectintel-anon, 2020-08-06:
> I'll reply to this in a bit, but holy shit that's hilarious.

LOL it really is

Ravenslofty, 2020-08-06:
So, putting my serious hat on, which version of ckb-next are you using? We very recently replaced AppIndicator with KStatusNotifier and if you're using a build older than two days ago or so, it might have already been fixed.

projectintel-anon, 2020-08-06:
> So, putting my serious hat on, which version of ckb-next are you using? We very recently replaced AppIndicator with KStatusNotifier and if you're using a build older than two days ago or so, it might have already been fixed.

I am using the most recent, I literally cloned and installed it not even 30 minutes ago.

Ravenslofty, 2020-08-06:
Okay, let's try *un*fixing it then: can you checkout 7d7cae92cb472994951ea059f08a22baf056c089 and see if that version (before the KStatusNotifier change) is a reasonable size?

projectintel-anon, 2020-08-06:
> Okay, let's try *un*fixing it then: can you checkout 7d7cae92cb472994951ea059f08a22baf056c089 and see if that version (before the KStatusNotifier change) is a reasonable size?

Okay call me dumb what am I doing exactly? All I see there is a code..

Ravenslofty, 2020-08-06:
`git checkout 7d7cae92cb472994951ea059f08a22baf056c089` and then build from source as before. To go back afterwards, use `git checkout master` and it'll automatically go back to the latest commit.

projectintel-anon, 2020-08-06:
That fixed it! Thanks!

Ravenslofty, 2020-08-06:
Well, kind of. It means that KStatusNotifier somehow isn't resizing things.

projectintel-anon, 2020-08-06:
Yeah :( using the older version you mentioned works though, however it hasn't fixed it properly for absolutely everybody. This certainly works for now however.

Ravenslofty, 2020-08-06:
I mean, I'm inclined to treat this as a bug in GNOME, but ugh. This isn't fun. Until we get it sorted you're probably going to be stuck on that version for a little while.

Ravenslofty, 2020-08-06:
Alternatively, there should be a checkbox in the settings menu which lets you disable the tray icon.

projectintel-anon, 2020-08-06:
I am okay with being on this version for the time being but yeah, that sucks. Keep up the amazing work, because you are doing fantastic. Love it!

projectintel-anon, 2020-08-06:
> Alternatively, there should be a checkbox in the settings menu which lets you disable the tray icon.

Yeah I thought there would be one, however I cannot get to it due to the sheer size of that tray icon!

tatokis, 2020-08-07:
Issue should now be fixed
https://github.com/ckb-next/ckb-next/commit/205b66e420ce89a666c5bcb92c4270c5503d3403

projectintel-anon, 2020-08-07:
> Issue should now be fixed
> https://github.com/ckb-next/ckb-next/commit/205b66e420ce89a666c5bcb92c4270c5503d3403

Nice!! Thanks!!

closed 2020-08-07 by Ravenslofty