ckb-next/ckb-next #631: K63 Wireless, new edition, Corsair protocol update

internalc, 2020-08-09:
Corsair K63 Wireless gaming keyboard

Introduction:

There seems to be a second edition of the keyboard, with new ID's and some protocol changes.
Existing K63 Wireless ID's found on **wireless** branch:
```
#define P_K63_NRGB_WL        0x1b45 /* wireless */
#define P_K63_NRGB_WL2       0x1b50 /* wireless */
```
while mine is:
```
#define P_K63_NRGB_WL3       0x1b8c /* wireless - usb cable*/
#define P_K63_NRGB_WL4       0x1b8f /* wireless - usb dongle*/
```

While the already provided K63 support mostly works, the lightning is broken on my unit. If I'm not wrong (looking quickly through the code) previous corsair protocol allowed to control the LEDs directly, which allowed to create any lightning effect. The new K63 Wireless has the lightning effect fixed in the firmware.

Here's the new lightning protocol I captured.

### 07 05 02 00 0X - Lightning intensity switch

There are 3 levels of lightning intensity, (provided that level 3 is not disabled, see `07 ac` command below).
Current ckb-next UI also list 3 levels so we can match it:
`X` = 0 - lightning off
`X` = 1 - 33 %
`X` = 2 - 67 %
`X` = 3 - 100 %

### 07 ac 00 00 0X - Lightning intensity limit 

When you toggle the light on K63 via the built-in lightning button the light intensity cycle through 0 (off), 1, 2, 3 levels.
The level 3 is disabled by default. In iCUE it's controlled by "Enable 100% light intensity" checkbox.
`X` = 1 - enable level 3
`X` = 0 - disable level 3

### 07 aa 00 00 ZZ 0X 0R YY- Lightning effects

`ZZ` - lightning effect
*The names of the effects taken from iCUE.
`ZZ` = ff - `NO EFFECT`, the lights are off and can't be turned on by the hardware button on the keyboard
`ZZ` = 01 - `PULSE`
`ZZ` = 04 - `WAVE`
`ZZ` = 05 - `VISOR`
`ZZ` = 06 - `RAIN`
`ZZ` = 07 - `STATIC COLOUR`
`ZZ` = 08 - `TYPE LIGHTNING KEY`
`ZZ` = 08 - `TYPE LIGHTNING RIPPLE`

`X` - lightning effect animation speed (1,2,3)

`R` - lightning effect animation direction, applicable only for `WAVE` effect
`R` = 0 - all effects except `WAVE`
`R` = 1 - left
`R` = 2 - right

`YY` - lightning opacity, linear 0 - 64

tatokis, 2020-08-09:
Thanks for this, however we are already aware of the new commands. No one got around to implementing them until now. The dark core shares mostly the same ones, but there's a byte difference in the packet.
You can have a look here https://github.com/ckb-next/ckb-next/pull/627/commits/5ae8dff9a6e67141b6de65b9df35cdbcd99d4fa0 and the wireless branch.

Not sure if K63 support is wired in for the hwanim command yet. You'll also probably have to manually echo the commands to /dev/input/ckbX/cmd

closed 2020-08-09 by tatokis

Ravenslofty, 2020-08-09:
While this isn't really an issue with ckb-next, I think it would be a good idea to instead post the text of your issue in ckb-next/CueShark which is the script we use to parse the protocol, since this is a relevant piece of information for the protocol.

internalc, 2020-08-09:
@ZirconiumX @Ravenslofty 

PS. Next week I'm sending my keyboard for RMA as it is sometimes freezing even on Windows where only the `Power on whilst holding ESC` brings it back to life. I'll which one I get back, new or old IDs.

PS2. Yes, I'm already using the code from `wireless` branch. It's mostly working, which is great. I'm going to implement the missing pieces having some spare minutes. However, I'm not using the lightning feature myself ;-) I guess the first thing to do would be to provide a dedicated vtable for the keyboard to disable commands which doesn't work. For the time being, I thought it would be helpful to document what I know alreaday.

internalc, 2020-08-09:
@tatokis I would kindly request to reopen the issue. Thanks for pointing the implementation draft, however the first thing any newcomer reads is the documentation. I would really suggest to share as much as we know about the protocol first, then one day we can implement/fix it.

@ZirconiumX could you merge the **HW ANIM** protocol extension to the prodocol page on the Wiki?

Ravenslofty, 2020-08-09:
I am @ZirconiumX :P (e.g. https://github.com/ZirconiumX/prussia redirects to https://github.com/Ravenslofty/prussia)

Ravenslofty, 2020-08-09:
As for the documentation; it's out of date because nobody much cared about it, and I found it much easier to have a physical explanation in a packet trace of what is happening, which is why CueShark is much more up to date than the protocol docs.

internalc, 2020-08-09:
> As for the documentation; it's out of date because nobody much cared about it, and I found it much easier to have a physical explanation in a packet trace of what is happening, which is why CueShark is much more up to date than the protocol docs.

I like CueShark! I would suggest to mention it on the Wiki and perhaps remove the protocol description or clearly mark it as not maintained