ckb-next/ckb-next #633: Gnome tray icon bug

SkavenNoises, 2020-08-16:
Report:
[ckb-next-report-1597567094.gz](https://github.com/ckb-next/ckb-next/files/5080287/ckb-next-report-1597567094.gz)

## General information

### Source
- Package from AUR (Manjaro)

## What is the program's current behaviour?
Currently on launch of the app, be it auto-start or manually, the system tray icon is a garbled mess of pixels (See photo).
[image]

On-top of this, the toggles in settings>behaviour currently do not function as intended.
- Hide tray icon - does not persist through re-log. Cycling toggle will fix the icon, but only during current log-in
- Monochrome tray icon - does not produce a monochrome tray icon. toggling will restore the rainbow coloured icon

- Selecting either of these settings after login+launch will toggle the pixilated mess into the standard rainbow coloured icon

## What is the expected behaviour of the program?
Toggle behaviour to correct and to persist through log-in cycles

tatokis, 2020-08-16:
Can you quit ckb-next and then run it from a terminal and send the output? Ideally when the tray icon is broken as well.

tatokis, 2020-08-16:
Also if possible, provide logs from your desktop environment as well.

tatokis, 2020-08-16:
I was able to reproduce it in a VM, and I have pushed a workaround.
Please try again with the latest version.

SkavenNoises, 2020-08-17:
Can confirm, behaviour works as intended now. Thank you for such a swift fix!

closed 2020-08-17 by SkavenNoises