ckb-next/ckb-next #634: Question: VOID RGB ELITE Wireless Premium Gaming Headset support

t3kg33k, 2020-08-25:
First of all, I want to say thank you for your hard work on this. My K65 RGB works wonderfully.
Second, are there any plan to add support for other Corsair hardware, specifically the VOID RGB ELITE Wireless Premium Gaming Headset?
Thanks.

tatokis, 2020-08-25:
> **Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

Aside from that, it depends. The original void was really terrible to work with, and we never finished support for it.
If this one uses the new protocol, it might be a while, but ckb will probably gain support for it.

dragonhelios, 2020-08-25:
@tatokis thanks for all the work you all do,  can't wait for the day I can manage all my corsair devices from one application in Linux. 

I'm sure t3kg33k will still respond but thought I can just as well. Attached report for Void RGB Elite Wireless. As a side note, don't know if it will help with the work being done in bragi and/or wireless branches on the dark core but, report does include information for the Dark Core RGB Pro SE as well.

[ckb-next-report-1598389194.gz](https://github.com/ckb-next/ckb-next/files/5126283/ckb-next-report-1598389194.gz).

please let me know if you need help with any testing.

t3kg33k, 2020-08-25:
Yes. Sorry. Just got home about an hour ago after a 14 hour shift.
@tatokis Definitely let me know if there is anything I can assist with. I'm not a developer but if you want info from me on the device, I can provide. And, if you want me to test code, I'm here and available.

gravityfargo, 2023-04-21:
closing in favor of #900

closed 2023-04-21 by gravityfargo