ckb-next/ckb-next #635: Lenovo Legion 7i 15IMH05

rajo, 2020-09-08:
Hi there,

I have a Lenovo Legion 7i Laptop which comes with the iCue-Software on Windows. As I'm using Ubuntu, I'd love to use ckb-next, but it can't find any devices. On Windows three different things can be configured: the keyboard, a LED strip on the front, and the LED light in the vents.

So, I created a report here and tried to collect as much information as possible with the commands from the other threads. If some piece of information is missing, just tell and I will update it. I'd be happy, if I could adjust things also from linux (and especially if I could store settings in Hardware s.t. it sticks also during boot etc.)

[ckb-next-report-1599597953.gz](https://github.com/ckb-next/ckb-next/files/5190990/ckb-next-report-1599597953.gz)

```
root@Teror:/home/rainer# dmesg | grep -i usb
[127345.681704] usb 1-2: USB disconnect, device number 14
[130996.029231] usb 1-6: reset high-speed USB device number 3 using xhci_hcd
[130996.432505] usb usb3: root hub lost power or was reset
[130996.432506] usb usb4: root hub lost power or was reset
[131020.232745] usb 1-2: new high-speed USB device number 15 using xhci_hcd
[131020.441742] usb 1-2: New USB device found, idVendor=0402, idProduct=5621, bcdDevice= 1.09
[131020.441743] usb 1-2: New USB device strings: Mfr=0, Product=1, SerialNumber=2
[131020.441744] usb 1-2: Product: USB 2.0 Storage Device
[131020.441744] usb 1-2: SerialNumber: 02003222200000039546
[131020.445220] usb-storage 1-2:1.0: USB Mass Storage device detected
[131020.445572] scsi host2: usb-storage 1-2:1.0
[133386.397297] usb 1-2: USB disconnect, device number 15
[133403.893203] usb 1-2: new high-speed USB device number 16 using xhci_hcd
[133404.101894] usb 1-2: New USB device found, idVendor=0402, idProduct=5621, bcdDevice= 1.09
[133404.101896] usb 1-2: New USB device strings: Mfr=0, Product=1, SerialNumber=2
[133404.101898] usb 1-2: Product: USB 2.0 Storage Device
[133404.101899] usb 1-2: SerialNumber: 02003222200000039546
[133404.104767] usb-storage 1-2:1.0: USB Mass Storage device detected
[133404.107449] scsi host3: usb-storage 1-2:1.0
[138267.089127] input: ITE Tech. Inc. ITE Device(8296) Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:C956.0005/input/input31
[138267.151007] input: ITE Tech. Inc. ITE Device(8296) Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:C956.0005/input/input32
[138267.151074] input: ITE Tech. Inc. ITE Device(8296) Wireless Radio Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:C956.0005/input/input33
[138267.151116] input: ITE Tech. Inc. ITE Device(8296) System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:C956.0005/input/input34
[138267.151186] hid-generic 0003:048D:C956.0005: input,hiddev0,hidraw1: USB HID v1.10 Keyboard [ITE Tech. Inc. ITE Device(8296)] on usb-0000:00:14.0-4/input0
[138287.649522] input: PixArt Lenovo USB Optical Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7.2/1-7.2.4/1-7.2.4:1.0/0003:17EF:6019.0006/input/input35
[138287.708160] hid-generic 0003:17EF:6019.0006: input,hidraw0: USB HID v1.11 Mouse [PixArt Lenovo USB Optical Mouse] on usb-0000:00:14.0-7.2.4/input0
[138576.671263] usb 1-2: USB disconnect, device number 16
```

```
root@Teror:/home/rainer# lsusb 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 002 Device 003: ID 0411:02af BUFFALO INC. (formerly MelCo., Inc.) 
Bus 002 Device 002: ID 0bda:0423 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 17ef:6019 Lenovo ITE Device(8296)
Bus 001 Device 007: ID 046a:0011 Cherry GmbH G83 (RS 6000) Keyboard
Bus 001 Device 006: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 004: ID 0bda:5423 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 003: ID 174f:2435 Syntek Integrated Camera
Bus 001 Device 002: ID 048d:c956 Integrated Technology Express, Inc. ITE Device(8296)
Bus 001 Device 016: ID 0402:5621 ALi Corp. M5621 High-Speed IDE Controller
Bus 001 Device 005: ID 8087:0026 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
root@Teror:/home/rainer# usbhid-dump -m 048d:c956
001:002:000:DESCRIPTOR         1599597325.379668
 06 89 FF 09 10 A1 01 85 5A 09 01 15 00 26 FF 00
 75 08 95 10 B1 00 C0 06 89 FF 09 07 A1 01 85 07
 09 01 15 00 26 FF 00 75 08 95 BF B1 00 C0 06 89
 FF 09 CC A1 01 85 CC 09 01 15 00 26 FF 00 75 03
 95 20 B1 00 C0 05 01 09 06 A1 01 85 01 75 01 95
 08 05 07 19 E0 29 E7 15 00 25 01 81 02 95 01 75
 08 81 03 95 05 75 01 05 08 19 01 29 05 91 02 95
 01 75 03 91 03 05 07 19 00 2A FF 00 15 00 26 FF
 00 95 06 75 08 81 00 05 07 19 00 29 DF 15 00 25
 01 95 E0 75 01 81 02 C0 05 0C 09 01 A1 01 85 02
 19 00 2A 3C 02 15 00 26 3C 02 75 10 95 01 81 00
 C0 05 01 09 0C A1 01 85 03 15 00 25 01 09 C6 95
 01 75 01 81 06 75 07 81 03 C0 05 01 09 80 A1 01
 85 04 19 81 29 83 15 00 25 01 95 08 75 01 81 02
 C0
```

```
root@Teror:/home/rainer# usbhid-dump -m 17ef:6019
001:008:000:DESCRIPTOR         1599597345.940670
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 03
 15 00 25 01 95 08 75 01 81 02 05 01 09 30 09 31
 09 38 15 81 25 7F 75 08 95 03 81 06 C0 C0
```

```
root@Teror:/home/rainer# cat /sys/devices/virtual/dmi/id/product_name 
81YT
```

```
root@Teror:/home/rainer# xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ MSFT0001:02 06CB:CE2D Mouse             	id=18	[slave  pointer  (2)]
⎜   ↳ MSFT0001:02 06CB:CE2D Touchpad          	id=19	[slave  pointer  (2)]
⎜   ↳ ITE Tech. Inc. ITE Device(8296) Consumer Control	id=13	[slave  pointer  (2)]
⎜   ↳ PixArt Lenovo USB Optical Mouse         	id=17	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=10	[slave  keyboard (3)]
    ↳ Integrated Camera: Integrated C         	id=15	[slave  keyboard (3)]
    ↳ HID 046a:0011                           	id=16	[slave  keyboard (3)]
    ↳ Ideapad extra buttons                   	id=20	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=21	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ ITE Tech. Inc. ITE Device(8296) System Control	id=11	[slave  keyboard (3)]
    ↳ ITE Tech. Inc. ITE Device(8296) Keyboard	id=12	[slave  keyboard (3)]
    ↳ ITE Tech. Inc. ITE Device(8296) Consumer Control	id=14	[slave  keyboard (3)]
    ↳ ITE Tech. Inc. ITE Device(8296) Wireless Radio Control	id=22	[slave  keyboard (3)]
```

```
root@Teror:/home/rainer# find /dev/input/ckb* -print -type f -exec cat {} \;
/dev/input/ckb0
/dev/input/ckb0/pid
437556
/dev/input/ckb0/version
0.4.2
/dev/input/ckb0/connected
```

tatokis, 2020-09-08:
These devices have their own protocol. I haven't finished implementing it and I got sidetracked by other things.
You can have a look at https://github.com/ckb-next/ckb-next/issues/330 and see if modifying the USB ID in the `legion` branch and compiling the code could get you anywhere. 
It won't show up in the GUI, but there are instructions in that issue on how to get it to do something from the command line.

rajo, 2020-09-09:
Thanks for the hint. Checket out the legion branch, added an additional USB ID and compiled. Now I get this output

```
software/ckb-next# ckb-next-daemon
ckb-next: Corsair RGB driver 0.4.2-25-g9e2ae63
[I] Root controller ready at /dev/input/ckb0
[I] Connecting ITE Device(8296) at /dev/input/ckb1
[I] Device is in BIOS mode
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1

```

...what does that mean? :) So, obviously, something is found but what is bios-mode and how can I disable it? (Sorry for being uninformed here). The USB ID seems to be the correct one, as usbhid-dump -d 048d:c956 -e stream ist responding to keystrokes.

Can you give me another pointer where to dig deeper?

tatokis, 2020-09-10:
Corsair devices (which this is not), have a BIOS mode. Yours doesn't, but the daemon thinks it's in that mode.
Add the device in IS_SINGLE_EP(kb) in usb.h. I'd suggest, for testing, to replace the existing legion id instead of adding a new one.

> usbhid-dump -d 048d:c956 -e stream ist responding to keystrokes
I don't remember the details, but by the sounds of that, if the daemon claims the device it will probably break input as it doesn't know how to handle it. Until that can be sorted out, have an external keyboard handy in case something goes wrong. Stopping the daemon (with sigterm, not sigkill) will restore input in such an event.

raagn08, 2020-12-06:
Hey @rajo, wondering if you were able to get this working somehow? I recently got the Legion 7i, would love to have a solution to setup custom RGB while on linux.

raagn08, 2021-02-28:
hey @tatokis, I changed the existing legion id 0xc956 but that resulted in daemon detecting it in BIOS Mode. So as per your above comment, I added device in IS_SINGLE_EP(kb). It worked, well kind of. 
The input broke, I tried changing the RGB via these commands and it worked partially (that might be due to incorrect key mapping being used from a Y730 config.
Photo: https://imgur.com/a/jT170PI

> `echo active > /dev/input/ckb1/cmd` (assuming ckb1 is the keyboard) which should switch it to software mode.
> `echo rgb ff00ff > /dev/input/ckb1/cmd` should set the colour for most keys.
> `echo idle > /dev/input/ckb1/cmd` should revert the keyboard back to hardware mode.

Here is the output from usbhid-dump -d 048d:c956 -e stream for keys a,b,c

```
Starting dumping interrupt transfer stream
with 1 minute timeout.

001:003:000:STREAM             1614551335.236071
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:003:000:STREAM             1614551335.237070
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:003:000:STREAM             1614551336.847567
 01 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:003:000:STREAM             1614551336.919164
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:003:000:STREAM             1614551336.920537
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:003:000:STREAM             1614551338.117267
 01 00 00 05 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:003:000:STREAM             1614551338.189491
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:003:000:STREAM             1614551338.190598
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:003:000:STREAM             1614551339.136804
 01 00 00 06 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:003:000:STREAM             1614551339.205514
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:003:000:STREAM             1614551339.206513
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00
```

Not sure how to proceed from here