ckb-next/ckb-next #637: Building ckb-next from mrdev024 fails with error

thespyboss, 2020-09-18:
## General information

### Source
How and where did you get this program, e.g.:
Built manually from ./quickinstall
Package cloned from https://github.com/mrdev023/ckb-next.git

## What is the program's current behaviour?
Fails at 49% with error
## What is the expected behaviour of the program?
I have all the dependancies installed and this had successfully installed on another computer with Manjaro Budgie
## Any other notes & comments?
Look at image inserted. that is the error i am getting
[image]
[image]

OrsanZ, 2020-09-18:
I've ran into this issue too. You should be able to fix it by rebasing whatever branch your want to compile onto ckb-next/master.

thespyboss, 2020-09-18:
@OrsanZ How do i do that? Im not the best with git and using git. Can u give like a basic description on how to?

thespyboss, 2020-09-18:
Also after doing some research, This is being installed from another repo. This isnt from a branch in ckb-next. Its from mrdev024's repo and i can't rebase the ckb-next:master into mrdev024:master.

tatokis, 2020-09-18:
Until I can get all the branches rebased, you can apply this patch on top.
https://github.com/ckb-next/ckb-next/commit/a9f41cd8b8f5b04c0c66c6d94f96a9725943831e.patch

closed 2020-09-18 by tatokis