ckb-next/ckb-next #638: Corsair H115i Platinum (1b1c:0c17)

andrejpodzimek, 2020-09-23:
`ckb-next-dev-detect` apparently considers the device relevant, but it doesn't seem to be detected by the daemon and doesn't appear in the GUI either. Here's a report: [ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/5265529/ckb-next-dev-detect-report.gz)

I'm using [`ckb-next-git`](https://aur.archlinux.org/packages/ckb-next-git/), [this version](https://github.com/ckb-next/ckb-next/commit/7abbe536dfe9fdcaf2c12a84811ccc3cd276823a) in particular.

The Platinum cooler is already supported by [liquidctl](https://github.com/jonasmalacofilho/liquidctl) (which can also control its fan speeds), so this is not really a big issue. Unified support for more Corsair devices would be awesome though.

OrsanZ, 2020-09-23:
IIRC this uses a completely different protocol (specifically Aseteks), that has very limited documentation.

gravityfargo, 2023-04-21:
Note for the future, this AIO doesn't rely on the Commander controller.

gravityfargo, 2023-09-18:
I've added this specific device to the wiki. In terms of support, you'll need to watch the request for the commander core. You can subscribe to this issue https://github.com/ckb-next/ckb-next/issues/833