ckb-next/ckb-next #642: Lighting on the "Fn" key.

RBalconi, 2020-09-29:
First of all thanks for having continued with the software.
I have a K65 RGB and the Fn key is not being illuminated, it is 100% off. even in the software being configured to illuminate, I believe it is not a problem with my keyboard, as I tested it on Windows and it worked fine.

[image]

Thank you!

tatokis, 2020-09-29:
Add the ckb-next-git PPA https://launchpad.net/~tatokis/+archive/ubuntu/ckb-next-git and upgrade to the latest version

RBalconi, 2020-09-29:
> Add the ckb-next-git PPA https://launchpad.net/~tatokis/+archive/ubuntu/ckb-next-git and upgrade to the latest version

Was it the one I downloaded, take a look, is it the latest version?

[image]

tatokis, 2020-09-29:
Please generate a report and attach it to the issue

MuhClaren, 2021-05-02:
Using Linux Mint 19 (Cinnamon), I had this issue on my K65 RGB (CH-9000072-NA). I fixed it by editing ckb-next/src/gui/keymap.cpp lines 689, 690, 691. I just commented them out as so:
```
        //map.remove("rwin");
        //map["fn"] = KStrafeKeys[3];
        //map["fn"].x -= 12;
```
[image]

After I saved the edit, I clean-installed ckb-next from the terminal:
```
cd ckb-next/
./quickinstall
```

Once I launched ckb-next, I changed all the keys to a solid color, saved it to the device, and rebooted. The fn key lights up now as expected.

Your Mileage May Vary. No guarantee this will work for you.

tatokis, 2022-10-03:
@MuhClaren Is this still an issue? (I assume it is.)
Can you please share the usb id of your device (`lsusb -d1b1c:`)?

Apologies, this would've been a simple fix and it fell through the cracks...

MuhClaren, 2022-10-05:
> @MuhClaren Is this still an issue? (I assume it is.) Can you please share the usb id of your device (`lsusb -d1b1c:`)?
> 
> Apologies, this would've been a simple fix and it fell through the cracks...

Yes, as far as I know the issue persists in the current release.  Thanks for having a look at it. Here is the output you requested:

`Bus 003 Device 007: ID 1b1c:1b2e Corsair 
Bus 003 Device 006: ID 1b1c:1b17 Corsair `

Cheers.

tatokis, 2022-10-06:
Can you please give the code in the [k65-fn](https://github.com/ckb-next/ckb-next/tree/k65-fn) branch a try?

tatokis, 2022-10-12:
@MuhClaren Most likely, your GUI version does not match the daemon.

commit eabd72e9 (2022-10-12): Merge branch 'k65-fn'

Closes #642
closed 2022-10-12 by tatokis