ckb-next/ckb-next #643: Firmware won't download

mishugashu, 2020-09-29:
## General information

### Source
How and where did you get this program, e.g.:
AUR (git version)

## Logs & crash reports
[ckb-next-report-1601407988.gz](https://github.com/ckb-next/ckb-next/files/5301450/ckb-next-report-1601407988.gz)

## What is the program's current behaviour?
firmware won't download
[image]

## What is the expected behaviour of the program?
Download firmware

## Any other notes & comments?
Starting the GUI also seems to freeze keyboard input (captured in a different issue #604 ), but replugging the keyboard physically or restarting the daemon seem to work.

tatokis, 2020-10-07:
Wow. You have a really old firmware version of the K55. The reason input is broken is because when Corsair initially released it, they updated the firmware to the latest version relatively quickly, so we never got the chance to get ckb working with the old firmware versions. It's probably not worth it fixing that as it'll work when you update.

Now, the reason the firmware update is failing is because we haven't been up to date with corsair's new firmware, so it's failing verification. I'll fix it shortly and you should be able to update.
That said, if you don't mind, join us on IRC as we would like to archive that old firmware since it's not available anywhere, before you update.

mishugashu, 2020-10-07:
We fixed it in IRC.

Ended up having to install an old version of CUE on a Windows machine to update it out of the current firmware, then ckb-next took over from there!

closed 2020-10-07 by mishugashu