ckb-next/ckb-next #644: [not an issue] just wanted to thank all of you (and a question)

atomGit, 2020-10-02:
it's great to see individuals pick up the balls that the manufacturers drop

i'm running an M65 RGB Elite and the software seems to working fine - thanks to all of you who contribute to this project :kiss: 

in the wiki...
> * = hardware playback not supported. Settings will be saved to software only.

what does this mean? i'm guessing this means that settings are not stored in device memory?

Ravenslofty, 2020-10-02:
Hi, it's really heartwarming to get an email like this, and I'm really
grateful for it (I'm sure Tasos will reply himself).

As for "hardware playback"; your Corsair device can store animated profile
data in its internal memory, and play them back without ckb-next running.
We currently haven't reverse engineered this part of the protocol enough to
implement this, so for now it's disabled (and the "save to hardware" button
is greyed out). We do plan to implement it at some point though.

atomGit, 2020-10-02:
> it's really heartwarming to get an email like this, ...

too often developers only hear gripes and so i just wanted to let you guys know that your work is appreciated

> As for "hardware playback"; your Corsair device can store animated profile
data in its internal memory ...

got it - that's of no interest to me, however i am wondering if it's possible to store the config on the mouse so the demon doesn't need to run?

Ravenslofty, 2020-10-02:
> i am wondering if it's possible to store the config on the mouse so the demon doesn't need to run?

This is what "hardware playback" is: the mouse itself is doing the config, not ckb-next.

atomGit, 2020-10-02:
thanks

i'll let you guys close this as you see fit

tatokis, 2020-10-07:
Thank you for the kind words. Really appreciated.
Unfortunately, hardware playback is quite low on the list of to-dos.
I will be receiving some hardware that might help the process but it will be quite a while.

I will be closing this issue as there's already another issue and PR for hardware playback.

Thank you again!

closed 2020-10-07 by tatokis