ckb-next/ckb-next #645: M65 - swap profiles while button is pressed

atomGit, 2020-10-02:
M65 RBG Elite
ckb-next 0.4.2 (AUR)

is it possible to swap profiles only while a button is pressed?
on the M65 i'd like to swap profiles while the sniper button is pressed - i didn't see anything in issues or the wiki that explains this

on the 'binding' tab -> 'special' one can use a button to switch profiles and there's a 'wrap around' option which i expected to switch to the next/previous profile on each button press, however that isn't happening

tatokis, 2020-10-07:
>  on the M65 i'd like to swap profiles while the sniper button is pressed - i didn't see anything in issues or the wiki that explains this

I don't think that's currently possible, and it would be difficult to implement as it is right now.

> on the 'binding' tab -> 'special' one can use a button to switch profiles and there's a 'wrap around' option which i expected to switch to the next/previous profile on each button press, however that isn't happening

You would need to bind it to that to all the modes/profiles you're cycling through. The moment you switch from one profile to another, the binding of the other one takes effect, so it no longer cycles between profiles if it's not set.
I think that's what you meant.

atomGit, 2020-10-07:
> You would need to bind it to that to all the modes/profiles you're cycling through. 

the problem is that i wasn't able to switch at all - in other words, i could go from profile 1 to profile 2, much less back again

i'm going to close this since i programmed the on-board memory with the iCUE s/w running on a VM - unfortunately, it too has the same limitations in that one cannot remap buttons while a button is pressed, nor can you bind a button to switch profiles

closed 2020-10-07 by atomGit