ckb-next/ckb-next #646: ckb next gui is not starting due to missing shared library

pakro, 2020-10-09:
### General information

The UI won't start anymore. The daemon is working.
Don't know exactly when the issue appeared, as I just installed a new kernel and rebooted.

### Source
How and where did you get this program, e.g.:
  * AUR ckb-next-git-1:0.4.2.r112.ge0d3cd5a-1

## Logs & crash reports

ckb-next: error while loading shared libraries: libquazip5.so.1: cannot open shared object file: No such file or directory

mikeyjones, 2020-10-09:
I think the issue seems to be libzip5 was removed is a recent update to Arch. The error I am seeing is:

`ckb-next: error while loading shared libraries: libquazip5.so.1: cannot open shared object file: No such file or directory`

mikeyjones, 2020-10-09:
@pakro  I just had a quick look to figure out the issue, it looks like in a recent update of quazip the library files were renamed from libquazip5 to libquazip1-qt5. What worked for my was creating a syn link from the new file to the old file it was looking for, so if you run this in a terminal, ckb-next should start working again: 

`sudo ln -s /usr/lib/libquazip1-qt5.so.1.0 /usr/lib/libquazip5.so.1`

This should be a temporary fix until we can get an updated version ckb-next.

tatokis, 2020-10-09:
> This should be a temporary fix until we can get an updated version ckb-next.

There's nothing to do on our side. You have to rebuild the package on your end and it will link against the correct version.

closed 2020-10-09 by tatokis

pakro, 2020-10-09:
Ok I've tried rebuilding but the build seems to  be cached or at least some parts of it. I will clear the cache and recompile it. Thank you kindly for the help!

Retrograde-i486, 2020-10-09:
> @pakro I just had a quick look to figure out the issue, it looks like in a recent update of quazip the library files were renamed from libquazip5 to libquazip1-qt5. What worked for my was creating a [...]

Thank you sir, confirming that this workaround works

coma-toast, 2021-12-06:
Similar issue on Fedora 35 - I was able to get it to start by installing this:
[http://rpmfind.net/linux/opensuse/distribution/leap/15.2/repo/oss/x86_64/libquazip5-1-0.9-lp152.1.1.x86_64.rpm](http://rpmfind.net/linux/opensuse/distribution/leap/15.2/repo/oss/x86_64/libquazip5-1-0.9-lp152.1.1.x86_64.rpm)