ckb-next/ckb-next #647: LS100 Support

xDazlD, 2020-10-09:
# Before you proceed

- Check if your device is listed in the [wiki](https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware).
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

---

Enter the name of your device here and attach the report generated above.

LS100 (Smart Lighting Strip)

The generate report button is broken for me. I'm running Fedora 32 with ckb-next 0.4.2.
[image]
Running the script by downloading it from the repo works though.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/5356915/ckb-next-dev-detect-report.gz)

Elyhaka, 2021-03-01:
Hello, I own this device and I was wondering if I could help implement the driver for it. Do you have some documentation to get started ?

PrimAvalou, 2023-01-14:
Is the integration of this device planned for the future ?
Thanks !