ckb-next/ckb-next #648: Quazip shared library file changed name after 1.0, ckb-next doesn't start anymore

selplacei, 2020-10-10:
The application does not start, giving the following error in the terminal:

`ckb-next: error while loading shared libraries: libquazip5.so.1: cannot open shared object file: No such file or directory`

The error started appearing after upgrading `quazip` from 0.9.1 to 1.0 earlier today (Oct 9) and restarting the PC. ckb-next starts normally and works fine if a new symlink from `libquazip1-qt5.so` to `libquazip5.so.1` is created in `/usr/lib`.

This happens on Arch Linux, using the `ckb-next` package from the AUR. `quazip` is from the official `extra` repository.

Kyriella, 2020-10-10:
Same issue here with the same circumstances (Arch Linux, etc.) and the posted solution worked too. Thanks for the tip!

wylel, 2020-10-10:
Same here. However, the solution posted works for making it start, it does not detect my device anymore.

tatokis, 2020-10-10:
https://github.com/ckb-next/ckb-next/issues/646

closed 2020-10-10 by tatokis

Rysher, 2020-10-11:
I can confirm this issue, exact same circumstance (arch linux, ckb installed from AUR, problem started after upgrading packages), and I can also confirm the solution that @selplacei offered.
However, i am not clear on why this was marked as duplicate, and if so, why wasn't the duplicate issue linked?

Regardless, if anyone else encounters this issue, here is a layman's guide to the solution
``
cd /usr/lib
sudo ln -sf libquazip1-qt5.so.1.0 libquazip5.so.1
``

ckb should be successfully launched now

AaronBottegal, 2020-10-11:
It was commented directly above. It IS linked.

Rysher, 2020-10-11:
@AaronBottegal 
right! thank you! I feel stupid now for failing to notice it. my previous comment will be edited