ckb-next/ckb-next #649: CORSAIR K95 RGB PLATINUM XT

david-waterworth, 2020-10-19:
CORSAIR K95 RGB PLATINUM XT

[ckb-next-report-1603145228.gz](https://github.com/ckb-next/ckb-next/files/5405102/ckb-next-report-1603145228.gz)

* I think this is different from the supported CORSAIR K95 RGB PLATINUM but I'm not sure - if it's not, it doesn't appear when I install the version of ckb-next from the Ubuntu repo

tatokis, 2020-10-19:
https://github.com/ckb-next/ckb-next/issues/543

closed 2020-10-19 by tatokis

david-waterworth, 2020-10-19:
Thanks - annoyingly unlike my old Steel Series which could be programmed in Windows and then remembered it's settings when I boot into Linux, the Corsair doesn't. Thankfully it at least has an option to turn off the LED's - I really wish they'd either support Linux or at least have a "Professional" default with just white LEDS.

FryDay, 2020-12-21:
> Thanks - annoyingly unlike my old Steel Series which could be programmed in Windows and then remembered it's settings when I boot into Linux, the Corsair doesn't. Thankfully it at least has an [...]

iCUE will let you remember settings as long as you configure HARDWARE settings. My issue with this method is that the "G" keys cannot be mapped to F13+. I've heard of people having luck with mapping them to foreign keys and mapping in linux that way, but I haven't tried it yet.

david-waterworth, 2021-03-26:
Thanks @FryDay, I'll try that. Although my other complaint about the K95 is the white isn't particularly bright anyway - much dimmer than my Steel Series, even my Das Keyboard is easier to read the keys and they're not lit up. I guess I should learn to type without looking at the keys...