ckb-next/ckb-next #651: K100 support

cajuncooks, 2020-10-21:
Corsair's new shiny object.

Here's the log generated by ckb-next-dev-detect: [ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/5417356/ckb-next-dev-detect-report.gz)

If there's anything I can do to help, let me know; I'm a vaguely competent developer sometimes.

tatokis, 2020-10-21:
Can you please install wireshark with usbcap (when prompted during the installation) on windows, and capture the keyboard being plugged in with CUE running?

Alternatively, if you want to do usb passthrough, you can use usbmon+wireshark on linux and pass the device through to a vm.

cajuncooks, 2020-10-21:
I will do that on my Windows desktop this afternoon, sure.

cajuncooks, 2020-10-21:
[usbpcap_corsair_k100.pcapng.gz](https://github.com/ckb-next/ckb-next/files/5419185/usbpcap_corsair_k100.pcapng.gz)

Ravenslofty, 2020-11-17:
It's Bragi, ugh. That's gonna be fun.

JesseRiemens, 2020-12-14:
I'm not sure if I am up for the task, but can I help in any way? I would love to see this implemented, and I might as well help.

banjahman, 2020-12-14:
same, ill offer my services however needed. I just got one of these and would love to be able to use the features

billygilbrech, 2021-02-25:
I just got one as well today, willing to help out here. What can I do?

stefanorossiti, 2021-02-26:
up

banjahman, 2021-03-12:
looks like we have at least 3-4 devs who are willing to put some time into this, how can we get started on this?

billygilbrech, 2021-03-12:
I can provide some captures if there is a guide on how to do it.

tatokis, 2021-03-12:
Can someone try the code from the k95-xt branch and change the usb device id in usb.h for the k95 platinum xt to the one for the k100?

It's possible that input will break, so make sure you have a backup keyboard.
The GUI won't recognise it, but the daemon should. Check if you can type with it.
After that, try `echo active > /dev/input/ckb1/cmd` afterwards and check if you can type again.

Also send the log from the daemon

bjowi, 2021-03-15:
Changed as follows:

> -#define P_K95_PLATINUM_XT    0x1b89 // Don't think this needs to be in the macro
> +#define P_K95_PLATINUM_XT    0x1b7c // Don't think this needs to be in the macro

After daemon restart tried to push enter, and it looked like that enter was repeated multiple times in the terminal, but it came in spurts, not evenly. Also all keys were blinking. No possibility to do the echo active thing.

Log is attached.
[log.txt](https://github.com/ckb-next/ckb-next/files/6142894/log.txt)

Jacobo0, 2021-04-01:
I'm new to this world, just installed Linux for the first time so I won't be much help, just wondering if there were still plans to get the K100 supported on this software?

mcrapet, 2021-06-26:
Thanks to KitsuWhooa (irc) for help. This is a quick & dirty hack.

Use new-devices-redux branch (commit 142621fa641f6c1110e7c334760e7dc14728cb4a).
Apply patch: https://gist.github.com/mcrapet/40f8000b59a0a6b720c6c0e947012bf2

Compile:
```
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Debug -DWITH_GUI=OFF -DDEBUG_USB_SEND=ON -DDEBUG_USB_RECV=ON -DDEBUG_USB_INPUT=ON ..
make
```

Launch:
```
./bin/ckb-next-daemon
```

daemon seems to launch correctly. However at each keystroke I got:
> [W] handle_nkro_key_input (keymap.c:661): Got unknown NKRO key press 112
> [I] ckb1 Input Recv: 01 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00

mcrapet, 2021-07-06:
Keyboard is available in two flavors:
- Cherry MX Speed
- Corsair OPX
Have no idea if this makes a difference for software.

mcrapet, 2021-07-25:
Help welcome, see initial crude support:
https://github.com/mcrapet/ckb-next/tree/k100-take1

MasterRoCcO, 2021-07-26:
Once you start using your git, the mouse stops working. Keyboard cannot be customized. But it is recognized in the software

freetz-ng-mod, 2021-08-09:
Keyboard is now recognized, but not all colors are displayed correctly. And besides, the number 7 no longer works with number pads

Red44, 2021-08-22:
My keyboard is somehow not recognized and do not know if it is because I have a KVM or my OPX flavour. 
Here is my log https://pastebin.com/dUcCvFuy

mcrapet, 2021-08-22:
@Red44 could you retry with my latest commit? (100-take2-2021.08.08 branch)

LakesideMiners, 2021-10-04:
any updates on this?
I tried installing @mcrapet 's commit
i assume it worked, like I have installed the main branch, and I assume I did not need to do anything speical for this one aside from the different repo. 
Device isnt dected.

tatokis, 2021-11-05:
I had some more time to work on this. Please try the latest code from the `k100` branch again.
Not all LEDs are added, but I want to get an idea of what works and what doesn't.

If the mapping is off, please tell me a few keys as an example to get an idea what the difference might be. For example "I set A to red but C turns red instead on the actual keyboard".

MasterRoCcO, 2021-11-05:
I have all the buttons on blue (# 000CFF)
[image]

tatokis, 2021-11-27:
For the lights that work, what is the light mapping like? Are they incorrect?

> If the mapping is off, please tell me a few keys as an example to get an idea what the difference might be. For example "I set A to red but C turns red instead on the actual keyboard".

T99Rots, 2021-12-03:
> For the lights that work, what is the light mapping like? Are they incorrect?
> 
> > If the mapping is off, please tell me a few keys as an example to get an idea what the difference might be. For example "I set A to red but C turns red instead on the actual keyboard".

I made a couple videos of the keyboard going through red, green and blue, key by key, from left to right from top to bottom in a sequential pattern, i hope those can be of use when mapping the keys, if necessary i can also edit them with a screen recording of ckb next synced up to it. Over all the red channel seems to be mostly mapped correctly, but green and blue are all over the place, also some keys stay partially on, and others stay partially off at all times, i've included some pictures

Videos:
[Blue](https://www.youtube.com/watch?v=Pk1EF4Q14zs)
[Green](https://youtu.be/JU3Ws7CLJqs)
[Red](https://youtu.be/d5g3e1IGR48)

Images:
[All keys red](https://i.imgur.com/N83T9eB.jpeg)
[All keys green](https://i.imgur.com/gK4bZUu.jpeg)
[All keys blue](https://i.imgur.com/gWToBDW.jpeg)
[All keys off](https://i.imgur.com/SoAZuhH.jpeg)

tatokis, 2021-12-04:
@T99Rots Thank you very much for the detailed response and the videos. It's really weird that green and blue are this off. I have no idea why this happens. I made some changes that might help with the stuck lights. Can you test the k100 branch again please?

T99Rots, 2021-12-05:
@tatokis I just pulled the k100 branch and did a quickinstall again, seems to still be the exact same keys that stay on

tatokis, 2021-12-05:
Can you send the output of ckb-next-daemon? Should be able to do so with something like `journalctl -u ckb-next-daemon --no-pager`

T99Rots, 2021-12-05:
This is everything from the first time i installed the k100 branch
```
dec 02 21:04:27 tim-workstation ckb-next-daemon[1328]: [I] Caught signal 15
dec 02 21:04:27 tim-workstation ckb-next-daemon[1328]: [I] Closing root controller
dec 02 21:04:27 tim-workstation ckb-next-daemon[1328]: [I] Removed device path /dev/input/ckb0
dec 02 21:04:27 tim-workstation systemd[1]: Stopping Corsair Keyboards and Mice Daemon...
dec 02 21:04:27 tim-workstation systemd[1]: ckb-next-daemon.service: Deactivated successfully.
dec 02 21:04:27 tim-workstation systemd[1]: Stopped Corsair Keyboards and Mice Daemon.
dec 02 21:04:28 tim-workstation systemd[1]: Started Corsair Keyboards and Mice Daemon.
dec 02 21:04:28 tim-workstation ckb-next-daemon[73638]: ckb-next: Corsair RGB driver 0.4.4-138-gf9809f3
dec 02 21:04:28 tim-workstation ckb-next-daemon[73638]: [I] Root controller ready at /dev/input/ckb0
dec 02 21:04:28 tim-workstation ckb-next-daemon[73638]: [I] Connecting CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard at /dev/input/ckb1
dec 02 21:04:28 tim-workstation ckb-next-daemon[73638]: [I] Starting input thread for /dev/input/ckb1
dec 02 21:04:28 tim-workstation ckb-next-daemon[73638]: [I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-4/1-4:1.0/ep_81
dec 02 21:04:28 tim-workstation ckb-next-daemon[73638]: [I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-4/1-4:1.1/ep_82
dec 02 21:04:28 tim-workstation ckb-next-daemon[73638]: [I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-4/1-4:1.2/ep_83
dec 02 21:04:28 tim-workstation ckb-next-daemon[73638]: [I] Setup finished for /dev/input/ckb1
dec 03 03:31:15 tim-workstation ckb-next-daemon[73638]: [I] Caught signal 15
dec 03 03:31:15 tim-workstation systemd[1]: Stopping Corsair Keyboards and Mice Daemon...
dec 03 03:31:15 tim-workstation ckb-next-daemon[73638]: [I] Disconnecting /dev/input/ckb1
dec 03 03:31:15 tim-workstation ckb-next-daemon[73638]: [I] Caught internal signal SIGUSR2
dec 03 03:31:15 tim-workstation ckb-next-daemon[73638]: [I] Stopping indicator thread for ckb1 (4)
dec 03 03:31:15 tim-workstation ckb-next-daemon[73638]: [I] Caught internal signal SIGUSR2
dec 03 03:31:15 tim-workstation ckb-next-daemon[73638]: [I] ckb1: Bragi poll thread shutting down due to 4 (Interrupted system call)
dec 03 03:31:15 tim-workstation ckb-next-daemon[73638]: [I] Stopping input thread for /dev/input/ckb1
dec 03 03:31:15 tim-workstation ckb-next-daemon[73638]: [I] Removed device path /dev/input/ckb1
dec 03 03:31:15 tim-workstation ckb-next-daemon[73638]: [I] Closing root controller
dec 03 03:31:15 tim-workstation ckb-next-daemon[73638]: [I] Removed device path /dev/input/ckb0
dec 03 03:31:15 tim-workstation systemd[1]: ckb-next-daemon.service: Deactivated successfully.
dec 03 03:31:15 tim-workstation systemd[1]: Stopped Corsair Keyboards and Mice Daemon.
dec 03 03:31:15 tim-workstation systemd[1]: ckb-next-daemon.service: Consumed 1.185s CPU time.
-- Boot 46c521463ef74916abf0f9f1e73da413 --
dec 03 14:17:17 tim-workstation systemd[1]: Started Corsair Keyboards and Mice Daemon.
dec 03 14:17:17 tim-workstation ckb-next-daemon[1306]: ckb-next: Corsair RGB driver 0.4.4-138-gf9809f3
dec 03 14:17:17 tim-workstation ckb-next-daemon[1306]: [I] Root controller ready at /dev/input/ckb0
dec 03 14:17:17 tim-workstation ckb-next-daemon[1306]: [I] Connecting CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard at /dev/input/ckb1
dec 03 14:17:17 tim-workstation ckb-next-daemon[1306]: [I] Starting input thread for /dev/input/ckb1
dec 03 14:17:17 tim-workstation ckb-next-daemon[1306]: [I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-4/1-4:1.0/ep_81
dec 03 14:17:17 tim-workstation ckb-next-daemon[1306]: [I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-4/1-4:1.1/ep_82
dec 03 14:17:17 tim-workstation ckb-next-daemon[1306]: [I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-4/1-4:1.2/ep_83
dec 03 14:17:17 tim-workstation ckb-next-daemon[1306]: [I] Setup finished for /dev/input/ckb1
dec 04 02:35:35 tim-workstation ckb-next-daemon[1306]: [I] Caught signal 15
dec 04 02:35:35 tim-workstation systemd[1]: Stopping Corsair Keyboards and Mice Daemon...
dec 04 02:35:35 tim-workstation ckb-next-daemon[1306]: [I] Disconnecting /dev/input/ckb1
dec 04 02:35:35 tim-workstation ckb-next-daemon[1306]: [I] Caught internal signal SIGUSR2
dec 04 02:35:35 tim-workstation ckb-next-daemon[1306]: [I] Stopping indicator thread for ckb1 (4)
dec 04 02:35:35 tim-workstation ckb-next-daemon[1306]: [I] Caught internal signal SIGUSR2
dec 04 02:35:35 tim-workstation ckb-next-daemon[1306]: [I] ckb1: Bragi poll thread shutting down due to 4 (Interrupted system call)
dec 04 02:35:35 tim-workstation ckb-next-daemon[1306]: [I] Stopping input thread for /dev/input/ckb1
dec 04 02:35:35 tim-workstation ckb-next-daemon[1306]: [I] Removed device path /dev/input/ckb1
dec 04 02:35:35 tim-workstation ckb-next-daemon[1306]: [I] Closing root controller
dec 04 02:35:35 tim-workstation ckb-next-daemon[1306]: [I] Removed device path /dev/input/ckb0
dec 04 02:35:35 tim-workstation systemd[1]: ckb-next-daemon.service: Deactivated successfully.
dec 04 02:35:35 tim-workstation systemd[1]: Stopped Corsair Keyboards and Mice Daemon.
[... cut ...]
dec 05 01:47:27 tim-workstation ckb-next-daemon[1329]: [I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-4/1-4:1.0/ep_81
dec 05 01:47:27 tim-workstation ckb-next-daemon[1329]: [I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-4/1-4:1.1/ep_82
dec 05 01:47:27 tim-workstation ckb-next-daemon[1329]: [I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-4/1-4:1.2/ep_83
dec 05 01:47:27 tim-workstation ckb-next-daemon[1329]: [I] Setup finished for /dev/input/ckb1
```

tatokis, 2021-12-07:
@T99Rots Can you try it again please?
Turns out that for some reason your k100 accepts the already implemented lighting packets just fine, which explains the odd behaviour.

T99Rots, 2021-12-08:
@tatokis I just tested it, apart from the leds in the ring button and the leds on the side of the keyboard which do not do not exist in the interface yet, everything is mapped correctly

fireball2k2000, 2022-02-07:
Hi everyone - at first: great work, I'm pleased that I can use my brand new K100 OPX under Linux as well.

I Installed the K100 branch on my Arch installation, which works great so far (despite the known problems with the wheel and the LEDs on the sides, which are only partly mapped as far is I can see), the "3" key (and also Fn, which is completely black) is also off on my keyboard, some kind of blue-greenish on completely white set. Any idea? Do you need information, captures, logs?

Best regards
Marcus

1n9i9c7om, 2022-03-31:
Hi everyone, first of all thanks for the great work and effort you're putting into this project!

Sadly, I'm unable to get this to work with my K100 at the moment. 

These are my logs: 
I think the interesting part is here, though:
`Mar 31 09:01:56.657397 nico-desktop-linux ckb-next-daemon[759]: [W] bragi_usb_read (via bragi_common.c:9): ckb1: Timeout while waiting for response
Mar 31 09:01:56.657397 nico-desktop-linux ckb-next-daemon[759]: [F] start_bragi_common (device_bragi.c:157): ckb1: Couldn't get bragi device mode. Aborting`

I've looked into device_bragi.c:157 and it seems like its attempting to switch into hardware mode, which does not work for some reason.

Any ideas what could be causing this issue?

cKouT, 2022-04-19:
hey everyone,
I'm really sorry but I'm a big noob in linux and i do not know how apply the patch. Can anyone help me pease.

Thanks

Grocel, 2022-06-06:
The polling rate being limited to 1000 Hz might me a general USB/Linux thing. It is unknown how 1000+ Hz works in Windows.

The Sabre Pro and the Sabre RGB Pro Championship Series mice are having similar characteristics. https://github.com/ckb-next/ckb-next/issues/709

Gaming hardware is getting ridiculous. I can't wait until we get polling rates of >1 MHz, so we can split atoms with our Mice and Keyboards. 🤭

MasterRoCcO, 2022-07-14:
Hi guys, I wanted to ask about the K100, will it also be supported when imckb-next v0.5.0? Because nothing has happened here for weeks.

dgcooke, 2022-07-19:
> Hi guys, I wanted to ask about the K100, will it also be supported when imckb-next v0.5.0? Because nothing has happened here for weeks.

I assume the answer is no.  I have 0.5.0 and a K100 which is unrecognized.

umop3plsdn, 2022-08-09:
ya i guess it was abandoned I just installed and  noticed no K100 in the list and came here to find this

MasterRoCcO, 2022-08-10:
git clone -b k100  https://github.com/ckb-next/ckb-next.git ckb-next_100

sudo apt install build-essential cmake libudev-dev qtbase5-dev zlib1g-dev libpulse-dev libquazip5-dev libqt5x11extras5-dev libxcb-screensaver0-dev libxcb-ewmh-dev libxcb1-dev qttools5-dev git libdbusmenu-qt5-dev
cd ckb-next_100
./quickinstall

tatokis, 2022-08-13:
I have pushed code to the [k100-v2](https://github.com/ckb-next/ckb-next/tree/k100-v2) branch, which should have all the lights fully functioning.
The zones on the light wheel are a bit broken at the moment, in the sense that the only way to select them is to click and drag the cursor in their general direction. The animations also render diagonally on them instead of vertically.
I'll try to fix them whenever I have time.

In addition, adding the extra wheel lights has increased the cpu usage while the preview is on.

I will also need someone to enable logging in the daemon (cmake -DCMAKE_BUILD_TYPE=Debug -DDEBUG_USB_INPUT=1) and then with the GUI running and brightness set to 0, move the control wheel while noting down the packets logged by the daemon so that I can implement support for it.

T99Rots, 2022-08-13:
@tatokis 

I just installed the branch and made a debug build for the logging
```
aug 13 12:33:05 tim-workstation ckb-next-daemon[81124]: [I] ckb1 Input: (83) 00 05 60 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
aug 13 12:33:10 tim-workstation ckb-next-daemon[81124]: [I] ckb1 Input: (83) 00 05 60 00 ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
The first one is rotating clockwise, 2nd one is counterclockwise

tatokis, 2022-10-04:
@T99Rots Thank you and apologies for the late reply. If you move the wheel as quickly as you can, do you get anything other than 01/ffffffff ?

Grocel, 2022-10-05:
> @tatokis I've tried spinning it as fast as i could and found out it's a great way to slice open your finger somehow

That's some AAA Product safety right there Corsair. Between this and paper cuts software developer is a deadly job.

cmarincia, 2022-10-16:
I own a K100 too, if you need any more info/testing I will help

m4xs31f3rt, 2022-11-18:
Just joined the K100 club and happily offering help debugging / programming. It's "hey man, long time no C!" for me but that won't stop me.

EnerGeh, 2022-12-17:
I recently got a K100 for a good deal at amazon and bought it without checking if its supported. I'm a longtime K95 user and addicted to your software, so I decided to join the discussion here.
After some investigation I found out that they have a running implementation at OpenRGB (https://openrgb.org/). They're using a GPL v2 license, so I think probably we can adopt some of their code?
As I'm the worst developer in the world, I assume that I've located their implementation here: https://github.com/CalcProgrammer1/OpenRGB/blob/master/Controllers/CorsairPeripheralController/CorsairK100Controller.cpp
May this be useful to make a step forward for the K100 support (or bragi in General)?

EnerGeh, 2023-05-03:
@Grocel is there a way to pay you a beer or two (or a coffee/tea/whatever you like)? Not to barge just to show you my admiration for that excellent piece of software :smiley:

Grocel, 2023-05-20:
Surely there is, but I am not the developer. 😅

mproffitt, 2023-06-28:
Hi. I recently bought the K100 optical mechanical gaming keyboard and have been trying to get this working using the `k100-v2` branch.

One thing I noticed is that this device has a different product ID (`0x1bc5`) from the two listed in this branch:
```
$ lsusb | grep K100
Bus 005 Device 002: ID 1b1c:1bc5 Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard
```

Not sure if this is related but this device is the Spanish keyboard layout with OPX keys.

Attempting to mirror the changes already made, I tested adding this device as `P_K100_GAMING` anticipating that the changes would result in similar working behaviour to what I anticipated from this thread and commit history for the two other devices.

What I see instead is the keyboard is barely usable (misses keystrokes, massively laggy, repeats itself a lot).

The daemon log being spammed with:

```
...
jun 28 16:57:16 nebula ckb-next-daemon[2354]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
jun 28 16:57:16 nebula ckb-next-daemon[2354]: [F] start_bragi_common (device_bragi.c:155): ckb1: Couldn't get bragi device mode. Aborting
...
```

Simultaneously, `dmesg` is spammed with:

```
...
[  128.103367] input: Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-1/5-1:1.0/0003:1B1C:1BC5.0124/input/input201
[  128.154578] hid-generic 0003:1B1C:1BC5.0124: input,hidraw2: USB HID v1.11 Keyboard [Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard] on usb-0000:0b:00.3-1/input0
[  128.154612] usb 5-1: usbfs: process 2834 (ckb1 input) did not claim interface 1 before use
[  128.229664] usb 5-1: usbfs: process 2834 (ckb1 input) did not claim interface 2 before use
[  128.229682] usb 5-1: usbfs: process 2834 (ckb1 input) did not claim interface 3 before use
[  128.430137] usb 5-1: usbfs: process 2834 (ckb1 input) did not claim interface 0 before use
[  128.548823] usb 5-1: reset high-speed USB device number 2 using xhci_hcd
...
``` 

Admittedly, I'm not much of a driver developer so almost certainly did something wrong whilst trying to emulate the device behaviour but I'm willing to experiment and help test this if someone can recognise the behaviour.

Not much of a windows guy either - but I have "Windows 11 to go" on a USB stick  and know my way around wireshark if additional packet captures are needed for this device as I'd like to see it included too :pray: 

[ckb-next-report-1687844499.gz](https://github.com/ckb-next/ckb-next/files/11897607/ckb-next-report-1687844499.gz)

d3st0y3r, 2023-11-09:
I regret buying this keyboard without first checking if it is compatible with Linux. However, I am hopeful that the developers of this project will provide support for the K100 in the future. I appreciate your efforts, guys.

xipinformatica, 2024-01-14:
I am also stuck with the k100 and CKB-Next not detecting it... It's been more than a year that I bought this Keyboard and I still can't use the RGB lighting on Linux... I wish I was a programer to make it compatible but I can just wait for you guys to add compatibility with this model.

DanySK, 2024-01-17:
Hi everyone, I've got a K100 `1b1c:1bc5 Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard` myself and I am willing to help. Let me know if there's anything I can help with.

Ahrkonh, 2024-02-01:
> Hi everyone, I've got a K100 `1b1c:1bc5 Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard` myself and I am willing to help. Let me know if there's anything I can help with.

Does ckb work with your model of keyboard? I ask because I have the 1bc5 k100 as well

mattTCC, 2024-02-01:
> > Hi everyone, I've got a K100 `1b1c:1bc5 Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard` myself and I am willing to help. Let me know if there's anything I can help with.
> 
> Does ckb work with your model of keyboard? I ask because I have the 1bc5 k100 as well

No not yet, that said you can still go to icue and add hardware lighting and keybindings if you have access to a windows machine

albinalm, 2024-03-13:
What's the status on this? :)

pyrotiger, 2024-10-27:
Just installed, encountering issue of no device detected. here is my report generated by CKB. 
[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/17533011/ckb-next-dev-detect-report.gz)
Report shows i have 1b7c for keyboard and my Dark Core SE is not being detected along with the Slipstream wireless dongle.

Rippeee, 2025-01-09:
Is there a way to get this keyboard detected?

pyrotiger, 2025-01-09:
Developer seems to have abandoned this project. Try OpenLinkHub

pyrotiger, 2025-04-02:
> OpenLinkHub supports less devices than this project currently. Only my
> keyboard is supported by OpenLinkHub, whereas ckb-next catches my mouse and
> Polaris mousepad too.

You can ask developer directly to add ur devices for support if you want.
He's very active in adding devices when he is able to for icue

Vash63, 2025-04-03:
> Developer seems to have abandoned this project. Try OpenLinkHub
> […](#)

https://github.com/ckb-next/ckb-next/commits/master/
This isn't what an abandoned project looks like, whether they have resources for this issue or not.

[1 more reply]

closed 2025-05-13 by gravityfargo