ckb-next/ckb-next #652: K95 RGB Illumination stays off after returning from susped

pakro, 2020-10-22:
## General information
K95 is working but the illumination is off and can't be turned on again.

### Source
How and where did you get this program, e.g.:
AUR: ckb-next-git

## Logs & crash reports
Attach:
[ckb-next-report-1603394366.gz](https://github.com/ckb-next/ckb-next/files/5425205/ckb-next-report-1603394366.gz)

## What is the program's current behaviour?
K95 illumination stays off after suspend.

## What is the expected behaviour of the program?
K95 illumination should be turned on after suspend.

## Any other notes & comments?
I've searched for an similar issue but weren't that lucky. However I've found something saying it is hard to detect because hardware and software mode of the keyboard.  I remember that this was working a couple of versions ago perfectly fine.

pakro, 2020-10-24:
After I've restarted the service the problem seems gone. Several wake from suspend now worked with the illumination turning on properly.

tatokis, 2020-10-24:
If this happens again, check dmesg.
On one of my computers, very rarely after resume, the kernel will throw an oops and break things.
I couldn't figure it out and I do not have access to that computer. It might be the same issue you are dealing with.