ckb-next/ckb-next #654: Switching modes permanently via command

heinihuber, 2020-10-24:
Hello ckb-next team, 

when I try to switch the mode via bash command, the target mode will be displayed only for a short moment but not permanently: the GUI is not opened while executing the command. If I e.g. switch from a profile that illuminates the keys in a dark blue color to a mode that should illuminate the keys in red color, the keyboard flashes one time in red and illuminates the keys in dark blue again afterwards.

What I'm doing wrong?

[ckb-next-report-1603544062.gz](https://github.com/ckb-next/ckb-next/files/5432905/ckb-next-report-1603544062.gz)

Command used to switch to the 2nd mode: **echo mode 2 switch > /dev/input/ckb1/cmd**

- OS: Ubuntu 20.x

- ckb-next installed via apt

- affected keyboard "K70 LUX RGB"

- firmware version 3.08

tatokis, 2020-10-24:
From the daemon manual:
> you can not use these commands with the GUI running at the same time.

Use `ckb-next --mode modename` instead. Or a combination of `--profile` and `--mode` to get the expected result.

tatokis, 2020-10-24:
Apologies, I completely misread your issue. How are you setting the lights on the mode? Just doing that isn't going to work on its own.

heinihuber, 2020-10-24:
I tried on different ways...

- after fresh reboot (option "Start ckb-next at login), GUI never opened before execution.

- after selecting a mode via the GUI and closing the GUI before executing the command.

- after using an rgb command (with GUI closed) e.g. "echo rgb 555555 > /dev/input/ckb1/cmd"

With the hint from you -> right syntax (**ckb-next --mode modename**) it works. Thank you for your help!

closed 2020-10-24 by heinihuber