ckb-next/ckb-next #658: Katar pro wireless

mossdonglonger, 2020-11-10:
# Before you proceed

- Check if your device is listed in the [wiki](https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware).
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

---
I have a Katar pro wireless and it is not detected both when connected with blue tooth and when connected with the 2.4ghz dongle. I installed ckb-next i think yesterday by building from source. Thanks
[ckb-next-report-1605043097.gz](https://github.com/ckb-next/ckb-next/files/5520227/ckb-next-report-1605043097.gz)

Daronfritts, 2022-12-31:
has support been added for the katar pro.  im new to github and linux for that matter so still figuring out how to navigate the both of them  thanks

GourSE, 2023-06-23:
> has support been added for the katar pro. im new to github and linux for that matter so still figuring out how to navigate the both of them thanks

No

Azurion42, 2023-07-17:
Throwing in my hat for requesting support, it's the mouse I use currently and would be nice to be able to use it properly.

gravityfargo, 2024-05-02:
Please plug the mouse in directly and rerun the report.

pepejw, 2024-06-26:
As in through 2.4GHz? There are no USB ports on the mouse itself.

[ckb-next-report-1719437618.gz](https://github.com/user-attachments/files/15996189/ckb-next-report-1719437618.gz)

MisterPrintf, 2024-09-12:
Would definitely appreciate support for this.

Report attached, generated immediately after boot.

[ckb-next-report-1726161046.gz](https://github.com/user-attachments/files/16983255/ckb-next-report-1726161046.gz)

Kosmixa, 2026-01-17:
> Looking for an update as I have a Katar Pro wireless as well. Not
> functioning in CKB-next

Lu1p90,

Please confirm if you are using the Katar Pro Wireless on a Linux system.
If so, could you also specify the distribution version you are using?
Wireless on Linux, and if so, which distro version?

Wekuz, 2026-01-17:
Not the person you asked it, but I also use the Katar Pro Wireless on Linux, more specifically Arch Linux and ckb-next shows "No devices connected"

lu1p90, 2026-01-17:
> Got it — this is a *very common Corsair-on-Linux issue*, and on Arch it’s
> usually one of *four concrete causes*. Let’s go through this *cleanly and
> methodically* so you can get your *Katar Pro Wireless* visible in *ckb-next*
> ------------------------------
> Important reality check (specific to Katar Pro Wireless)
> 
> ⚠️ *ckb-next does NOT support Corsair wireless devices over Bluetooth.*
> It *only works via the Corsair USB receiver (dongle)*.
> 
> If you’re connected via Bluetooth → *ckb-next will always say “No devices
> connected.”*
> 
> So first question (critical):
> 
>    -
> 
>    ✅ Are you using the *Corsair Slipstream USB receiver*?
> 
> If *yes*, continue below.
> If *no*, that’s the reason — and I’ll explain your options at the end.
> ------------------------------
> Step 1: Verify the USB receiver is actually detected
> 
> Plug in the Corsair USB dongle and run:
> 
> lsusb
> 
> You should see something like:
> 
> Bus XXX Device XXX: ID 1b1c:xxxx Corsair
> 
> If you *do not* see Corsair listed:
> 
>    -
> 
>    Try another USB port (preferably USB-A)
>    -
> 
>    Avoid USB hubs
>    -
> 
>    Confirm the dongle works on another system
> […](#)
> 
> ------------------------------
> Step 2: Install required packages (Arch-specific)
> 
> Make sure everything ckb-next needs is installed:
> 
> sudo pacman -S --needed ckb-next hidapi libusb qt6-base
> 
> If you installed from AUR:
> 
> paru -S ckb-next# or
> yay -S ckb-next
> 
> ------------------------------
> Step 3: Load required kernel modules
> 
> ckb-next requires raw HID access.
> 
> Run:
> 
> lsmod | grep hid
> 
> Ensure these exist:
> 
>    -
> 
>    hid_generic
>    -
> 
>    usbhid
> 
> If not:
> 
> sudo modprobe usbhidsudo modprobe hid_generic
> 
> ------------------------------
> Step 4: Fix permissions (this is usually the blocker)
> 
> ckb-next *cannot access HID devices without proper udev rules*.
> Install udev rules
> 
> Check if rules exist:
> 
> ls /usr/lib/udev/rules.d | grep corsair
> 
> If missing, create them:
> 
> sudo nano /etc/udev/rules.d/99-corsair.rules
> 
> Paste:
> 
> SUBSYSTEM=="usb", ATTR{idVendor}=="1b1c", MODE="0666"
> SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1b1c", MODE="0666"
> 
> Reload rules:
> 
> sudo udevadm control --reload-rulessudo udevadm trigger
> 
> Then *unplug and replug the dongle*.
> ------------------------------
> Step 5: Run ckb-next correctly
> 
> Start the daemon first:
> 
> sudo systemctl enable --now ckb-next-daemon
> 
> Then launch GUI:
> 
> ckb-next
> 
> ⚠️ *Do NOT run the GUI with sudo* — only the daemon needs root.
> ------------------------------
> Step 6: Confirm HID access
> 
> Run:
> 
> ls -l /dev/hidraw*
> 
> You should see permissions like:
> 
> crw-rw-rw-
> 
> If not, permissions are still wrong.
> ------------------------------
> Step 7: Debug mode (if still invisible)
> 
> Run:
> 
> ckb-next --debug
> 
> Or check logs:
> 
> journalctl -u ckb-next-daemon -e
> 
> If you see:
> 
>    -
> 
>    No supported devices found → receiver not recognized
>    -
> 
>    Permission denied → udev rules still incorrect
>    -
> 
>    Wireless device unsupported → expected (see below)
> 
> ------------------------------
> ⚠️ Hard Truth: Katar Pro Wireless support status
> 
> As of *2025*:
> Mode Works in ckb-next?
> USB Wired ❌ (mouse is wireless-only)
> USB Slipstream Receiver ⚠️ *Partial / flaky*
> Bluetooth ❌ Not supported
> 
> Corsair *does not publish wireless protocol specs*, so Linux support is
> reverse-engineered and incomplete.
> ------------------------------
> Best Workarounds (Realistic Options) ✅ Option 1: Use the dongle + basic
> functionality
> 
>    -
> 
>    DPI switching works (hardware)
>    -
> 
>    RGB usually defaults
>    -
> 
>    No software control
> 
> ✅ Option 2: Piper + libratbag (recommended)
> 
> Katar Pro Wireless works *better* here.
> 
> sudo pacman -S piper libratbagsudo systemctl enable --now ratbagd
> piper
> 
> ⚠️ DPI works, RGB may not.
> 
>  <[#658 (comment)](https://github.com/ckb-next/ckb-next/issues/658#issuecomment-3764255199)>
> 
>  Not the person you asked it, but I also use the Katar Pro Wireless on
>  Linux, more specifically Arch Linux and ckb-next shows "No devices
>  connected"
> 
>  <[#658 (comment)](https://github.com/ckb-next/ckb-next/issues/658#issuecomment-3764255199)>,
> 
> 
> Peace and Abundance
> Michael A. Canteen

Step 3 I'm getting an error:
❯ lsmod | grep hid
mac_hid                12288  0

~
❯ sudo modprobe usbhidsudo modprobe hid_generic
modprobe: FATAL: Module usbhidsudo not found in directory /lib/modules/6.18.5-2-cachyos

~

BrandtVanDenBerg, 2026-01-17:
> 1️⃣ The immediate error: command typo
> 
> You accidentally ran *both commands as one line*:
> 
> sudo modprobe usbhidsudo modprobe hid_generic
> 
> So the kernel tried to load a module named usbhidsudo — which obviously
> doesn’t exist.
> Correct commands (run separately):
> 
> sudo modprobe usbhidsudo modprobe hid_generic
> 
> ⚠️ Note: the module name is *usbhid*, not usbhids.
> ------------------------------
> 2️⃣ Why lsmod | grep hid only shows mac_hid
> 
> You’re on *CachyOS kernel 6.18.5*, which behaves slightly differently than
> vanilla Arch:
> 
>    -
> 
>    usbhid *autoloads only when a USB HID device is detected*
>    -
> 
>    If the Corsair receiver isn’t recognized as HID, the module never loads
>    -
> 
>    mac_hid being present is normal and irrelevant here (it’s harmless)
> 
> After fixing the typo, rerun:
> 
> sudo modprobe usbhidsudo modprobe hid_generic
> 
> Then verify:
> 
> lsmod | grep hid
> 
> You should now see at least:
> 
> usbhid
> hid_generic
> […](#)

Dude. Please stop. Please.

Nobody gives two shits about what ChatGPT thinks. ckb-next is fairly niche/specialized software in the grand scheme of things, and there is _zero_ way that a sufficient amount of data on this is included in its dataset for it to be remotely useful.

You are telling users to run root commands that you, likely, do not understand. Nor, evidently, does ChatGPT, since it is hallucinating kernel modules that are not present by defauly on CachyOS.

If we wanted help from a GPT model we could ask it ourselves. We do not need you to function as an intermediary; as a matter of fact, you come across condescendingly when you try to.

"If you don't got anything nice to say, don't say nothin' at all". That goes for _useful_ things as well.

lu1p90, 2026-01-17:
No luck here...

❯ sudo modprobe usbhid

~
❯ sudo modprobe hid_generic

~
❯ lsmod | grep hid
mac_hid                12288  0

~

Kosmixa, 2026-01-17:
> I understand root commands very well and was able to fix my system with no
> problems at all. Stop hating. If you're not going to provide a solution,
> then why say anything at all?

Then go right ahead and help fix the problem.

Kosmixa, 2026-01-17:
> *There is currently no solution* until support for this device is added.
> This is the *entire* point of this issue - it is a *feature request* for
> the device definitions to be added. Not a request for help because system
> breakage is going on.
> If you want to provide support to other users that is fine, ultimately -
> whether or not you give out erroneous information and/or break someone
> else's system is ultimately not my concern. But please take it elsewhere.

Ok!!

lu1p90, 2026-01-17:
Funny thing is I tried Piper as well and made sure rabagd is running and it also cannot detect the mouse...

lu1p90, 2026-01-17:
Went ahead a ordered a mouse from the supported list since I need to get some macros going. Sorry to cause a fuss lol