ckb-next/ckb-next #659: Not working with Strafe RGM on Arch Linux

mwknox186, 2020-11-12:
I've recently switched to linux and I couldn't get my keyboard to work.

REPORT:
ckb-next-dev-detect - Thu Nov 12 10:48:45 EST 2020

Kernel: Linux max-pc 5.9.6-arch1-1 #1 SMP PREEMPT Thu, 05 Nov 2020 21:00:46 +0000 x86_64 GNU/Linux

OS Version:
LSB Version:	1.4
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a

USB Devices:
Couldn't open device, some information will be missing

Bus 002 Device 008: ID 1b1c:1b20 Corsair STRAFE RGB Gaming Keyboard
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b20 STRAFE RGB Gaming Keyboard
  bcdDevice            2.05
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
[... 126 lines cut ...]
Nov 12 10:48:23.572286 max-pc ckb-next-daemon[92297]: [F] os_inputopen (input_linux.c:67): Failed to load uinput module
Nov 12 10:48:23.572286 max-pc ckb-next-daemon[92297]: [I] Disconnecting /dev/input/ckb1
Nov 12 10:48:24.924856 max-pc ckb-next-daemon[92297]: [I] Removed device path /dev/input/ckb1

Daemon on last boot:
-- Logs begin at Sat 2020-11-07 15:56:19 EST, end at Thu 2020-11-12 10:48:36 EST. --
-- No entries --

Dev nodes:
/dev/input/ckb0
/dev/input/ckb0/pid
92297
/dev/input/ckb0/version
0.4.2
/dev/input/ckb0/connected

Environment:
XDG_SESSION_DESKTOP=KDE
QT_BEARER_POLL_TIMEOUT=-1
XDG_CURRENT_DESKTOP=KDE
DISPLAY=:0
QT_AUTO_SCREEN_SCALE_FACTOR=0

Kernel command line:
BOOT_IMAGE=/vmlinuz-linux root=UUID=041304b7-2de8-461f-8aae-65eb0b57bc92 rw

Ravenslofty, 2020-11-12:
> Nov 12 10:48:23.572107 max-pc ckb-next-daemon[92641]: modprobe: FATAL: Module uinput not found in directory /lib/modules/5.9.6-arch1-1

Did you update your kernel recently? Try rebooting.

mwknox186, 2020-11-12:
That worked, Thanks!

closed 2020-11-12 by mwknox186