ckb-next/ckb-next #663: Qt5.15 removed a lot of APIs; no longer compiles.

egberts, 2020-11-18:
## General information
When working out of a fresh-installed Debian 10.5/bullseye/testing, it no longer compiles.

### Source
How and where did you get this program, e.g.:
  * git clone  #master 
  * Built manually using quickinstall script

## Logs & crash reports
Compile error:
```console
make
[ 19%] Built target ckb-next-daemon
[ 21%] Built target KissFFT
[ 22%] Built target pinwheel
[ 23%] Built target rain
[ 25%] Built target wave
[ 26%] Built target gradient
[ 27%] Built target ripple
[ 29%] Built target random
[ 31%] Built target heat
[ 32%] Built target invaders
[ 34%] Built target life
[ 36%] Built target snake
[ 38%] Built target pipe
[ 40%] Built target mviz
[ 42%] Built target ckb-next-lrelease
[ 43%] Automatic MOC and UIC for target ckb-next
[ 43%] Built target ckb-next_autogen
Scanning dependencies of target ckb-next
[ 43%] Building CXX object src/gui/CMakeFiles/ckb-next.dir/keywidget.cpp.o
/home/steve/admin/hardware/corsair/ckb-next/src/gui/keywidget.cpp:71:18: error: expected ')' before '*' token
   71 | void (QPaintEvent*){
      |      ~           ^
      |                  )
/home/steve/admin/hardware/corsair/ckb-next/src/gui/keywidget.cpp:13:163: warning: 'ironclawOverlay' defined but not used [-Wunused-variable]
   13 | , *harpOverlay = 0, *glaiveOverlay = 0, *polarisOverlay = 0, *katarOverlay = 0, *m95Overlay = 0, *ironclawOverlay = 0;
      |                                                                                                   ^~~~~~~~~~~~~~~
```

## What is the program's current behaviour?
Does not compile

## What is the expected behaviour of the program?
Should compile cleanly.

## Any other notes & comments?

egberts, 2020-11-18:
Looks like `QPaintEvent` is missing a typedef for `QPaintEvent` class or something.

```
#include <QPaintEvent>  // /usr/include/x86_64-linux-gnu/qt5/QtGui/QPaintEvent
   +-- qevent.h  // /usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h
        +  void (QPaintFresh *)  
```
I tested this by  inserting `#error it got here` before this `QPaintFresh` declaration and compiler picked that  up.  

Note
------
Identical error for both `gcc-10` and `clang-9`.

tatokis, 2020-11-19:
The function is missing its name on line 71 in keywidget.cpp on your local repo copy. It compiles just fine.

closed 2020-11-19 by tatokis