ckb-next/ckb-next #666: Upper edge lightbar (across entire keyboard) is not individually configurable

egberts, 2020-11-21:
The lightbar that spans across the top edge of Corsair K95 Platinum keyboard is not configurable to either intensity nor color.

[ckb-next-report-1605990106.gz](https://github.com/ckb-next/ckb-next/files/5578326/ckb-next-report-1605990106.gz)

## General information
I am using the #656 pull (ahead of master).

### Source
How and where did you get this program, e.g.:
  * Checked out d7bd30026a80a5b50e42e086e978ea55e8efd4bf from #520 and #656.
  * Built manually using quickinstall script

## Logs & crash reports

None

## What is the program's current behaviour?

Using the GUI of d7bd30026a80a5b50e42e086e978ea55e8efd4bf, the top-row entire-edge lightbar is not individually controlable.  

* We are able to give it a preprogrammed pattern
* Just not able to individually control each LED on the lightbar.

## What is the expected behaviour of the program?

## Any other notes & comments?

Ravenslofty, 2020-11-23:
You *can* individually change the LEDs on the lightbar (ckb-next's patterns just focus on lights). Just select the lights in the lighting tab.

egberts, 2020-12-19:
Ok.  I get this.   When selecting individual LED in the GUI, sometimes the colors don’t get updated or is hard to see such changes.  Perhaps a bit thicker line showing as over-exaggerated horizontal bar?

egberts, 2022-07-14:
This is just an aesthetic issue.  

I would like to make the top row in GUI a tad bit bigger as to let us know that they are CLICKABLE.  

Perhaps with a black line surrounding each clickable LED region. 

My two cents.

egberts, 2022-12-21:
lower non-functional esthetic priority.  GUI-issue.

egberts, 2023-01-21:
Closing this as originally fixed. 

Just a note that while it may be hard to see, it is in fact settable and changeable
.

closed 2023-01-21 by egberts