ckb-next/ckb-next #667: ckb-next-daemon service restart required after reboot.

Scary-Terrance, 2020-11-22:
This is using ckb-next installed from AUR on Manjaro with the latest kernel.   The settings I configure in the app do not apply automatically on restarting my pc.  In order to get proper settings I must restart the service with systemctl restart ckb-next-daemon after every reboot.  Everything functions correctly after a service restart.

Pillgar, 2020-11-22:
Did this happen recently after libxkbcommon-x11 v1.0.2 was updated?  If so, try downgrading it.  Post back if that solves it please.

MagiusCHE, 2021-01-13:
Same iussue here with "extra/libxkbcommon-x11 1.0.3-1" (Manjaro)
It worked until about 7 days ago but now after boot i need to manually restart "sudo systemctl restart ckb-next-daemon" in order to work again.
No error reported in "journalctl -u ckb-next-daemon".
Thx in advance.

lilongueti, 2021-01-29:
libxkbcommon-x11 1.0.3-1 (EndeavourOS )
A 'restart daemon' button under ckb-next settings would be great
dmesg after restarting the daemon
```

[    3.340578] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0002/input/input10
[    3.399604] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0002/input/input11
[    3.399655] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0002/input/input12
[    3.399697] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0002/input/input13
[    3.399763] hid-generic 0003:1B1C:1B3A.0002: input,hiddev97,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 RAPIDFIRE Keyboard ] on usb-0000:00:14.0-3.2/input0
[    3.402493] hid-generic 0003:1B1C:1B3A.0003: hiddev98,hidraw2: USB HID v1.11 Device [Corsair Corsair Gaming K70 RAPIDFIRE Keyboard ] on usb-0000:00:14.0-3.2/input1
[    3.439455] usb 1-9: new full-speed USB device number 8 using xhci_hcd
[    3.597133] usb 1-9: New USB device found, idVendor=048d, idProduct=c100, bcdDevice= 0.04
[    3.597135] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.597136] usb 1-9: Product: ITE Device(8910)
[    3.597137] usb 1-9: Manufacturer: ITE Tech. Inc.
[    3.608103] input: ITE Tech. Inc. ITE Device(8910) Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:048D:C100.0004/input/input21
[    4.073319] usb 1-3.3: New USB device found, idVendor=1b1c, idProduct=1b3e, bcdDevice= 3.14
[    4.073321] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.073322] usb 1-3.3: Product: Corsair Gaming SCIMITAR PRO RGB Mouse
[    4.073323] usb 1-3.3: Manufacturer: Corsair
[    4.073323] usb 1-3.3: SerialNumber: 0901F033AECC0C80570213F8F5001941
[    6.192950] input: Corsair Corsair Gaming SCIMITAR PRO RGB Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.3/1-3.3:1.0/0003:1B1C:1B3E.0005/input/input32
[    6.193095] corsair 0003:1B1C:1B3E.0005: input,hiddev97,hidraw1: USB HID v1.11 Mouse [Corsair Corsair Gaming SCIMITAR PRO RGB Mouse] on usb-0000:00:14.0-3.3/input0
[    6.193206] corsair 0003:1B1C:1B3E.0006: hiddev98,hidraw2: USB HID v1.11 Device [Corsair Corsair Gaming SCIMITAR PRO RGB Mouse] on usb-0000:00:14.0-3.3/input1
[    6.270616] usb 1-3.2: usbfs: process 511 (ckb1 input) did not claim interface 0 before use
[14312.826231] usb 1-3.2: usbfs: process 512 (ckb1 led) did not claim interface 0 before use
[14312.999340] usb 1-3.2: reset full-speed USB device number 7 using xhci_hcd
[14313.100496] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0007/input/input38
[14313.155955] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0007/input/input39
[14313.156015] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0007/input/input40
[14313.156065] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0007/input/input41
[14313.156158] hid-generic 0003:1B1C:1B3A.0007: input,hiddev97,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 RAPIDFIRE Keyboard ] on usb-0000:00:14.0-3.2/input0
[14313.158999] hid-generic 0003:1B1C:1B3A.0008: hiddev98,hidraw2: USB HID v1.11 Device [Corsair Corsair Gaming K70 RAPIDFIRE Keyboard ] on usb-0000:00:14.0-3.2/input1
[14313.287498] usb 1-3.1: 3:1: cannot get freq at ep 0x84
[14313.314584] usb 1-3.1: 3:1: cannot get freq at ep 0x84
[14313.336007] usb 1-3.2: usbfs: process 953 (ckb2 led) did not claim interface 0 before use
[14313.336138] usb 1-3.2: usbfs: process 952 (ckb2 input) did not claim interface 0 before use
[14313.509348] usb 1-3.2: reset full-speed USB device number 7 using xhci_hcd
[14313.613442] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0009/input/input42
[14313.672777] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0009/input/input43
[14313.673026] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0009/input/input44
[14313.673080] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0009/input/input45
[14313.673165] hid-generic 0003:1B1C:1B3A.0009: input,hiddev97,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 RAPIDFIRE Keyboard ] on usb-0000:00:14.0-3.2/input0
[14313.675946] hid-generic 0003:1B1C:1B3A.000A: hiddev98,hidraw2: USB HID v1.11 Device [Corsair Corsair Gaming K70 RAPIDFIRE Keyboard ] on usb-0000:00:14.0-3.2/input1
[14313.675974] usb 1-3.2: usbfs: process 512 (ckb1 led) did not claim interface 0 before use
[14313.946025] usb 1-3.2: usbfs: process 390 (ckb1 usb) did not claim interface 1 before use
[14314.122678] usb 1-3.2: reset full-speed USB device number 7 using xhci_hcd
[... cut ...]
[14371.603407] input: Corsair Corsair Gaming K70 RAPIDFIRE Keyboard  as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/0003:1B1C:1B3A.0012/input/input62
[14371.603791] hid-generic 0003:1B1C:1B3A.0012: input,hiddev100,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 RAPIDFIRE Keyboard ] on usb-0000:00:14.0-3.2/input0
[14371.607189] hid-generic 0003:1B1C:1B3A.0013: hiddev101,hidraw5: USB HID v1.11 Device [Corsair Corsair Gaming K70 RAPIDFIRE Keyboard ] on usb-0000:00:14.0-3.2/input1

```

TheTee82, 2021-02-04:
Similar issue here - however after boot the keyboard is (sometimes) unresponsive so in order to get it to work, I need to:
* unplug/replug keyboard
* restart `ckb-next-daemon` (since I'm not able to restart the daemon without keyboard

NOTE:
* when the issue occurs and I open ckb-next GUI, I noticed 2(!) identical tabs with my keyboard configs next to each other
  * I assume (and I might be wrong here), my keyboard is registered twice and thus the keyboard is unresponsive due to some kind of conflict with each other

---

**ckb-next version**: `ckb-next 0.4.3-7-ge48e6e6` (manually built directly from sources - however the versions provided in AUR have the same issue: https://aur.archlinux.org/packages/ckb-next / https://aur.archlinux.org/packages/ckb-next-git )

**System Infos** (at the time of a working keyboard):
```
inxi
CPU: 6-Core Intel Core i7-8700K (-MT MCP-) speed/min/max: 1083/800/4700 MHz Kernel: 5.10.7-3-MANJARO x86_64 Up: 24m 
Mem: 4374.2/32048.6 MiB (13.6%) Storage: 7.04 TiB (42.8% used) Procs: 359 Shell: Bash inxi: 3.2.02
```
```
lsusb -v 2>/dev/null | egrep '(^Bus|Keyboard)' | grep -B1 Keyboard
Bus 001 Device 007: ID 1b1c:1b55 Corsair CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming Keyboard
  iProduct                2 CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming Keyboard
      bInterfaceProtocol      1 Keyboard
```
```
libxkbcommon-x11 1.0.3-1
```

--- 

**Logs**:
* ckb-next generated report - generated after boot (keyboard not working): 
[ckb-next-report-1612422128.gz](https://github.com/ckb-next/ckb-next/files/5923939/ckb-next-report-1612422128.gz)
* `journalctl -u ckb-next-daemon` (after I replugged the keyboard): 
[journalctl_ckb-next-daemon.gz](https://github.com/ckb-next/ckb-next/files/5923943/journalctl_ckb-next-daemon.gz)

---

If you need anything else for debugging, please let me know, I'm happy to help.

tatokis, 2021-02-04:
@TheTee82 Based on the logs, it sounds like there's some application or another driver that's refusing to let go of the keyboard.
Try temporarily disabling the daemon `sudo systemctl disable ckb-next-daemon` and then after a reboot, run `usb-devices` and have a look at the block with the line similar to `P:  Vendor=1b1c ProdID=1b55`.
Specifically the `I:` lines below it should have the drivers listed.

When the daemon is disabled, the gui should prompt you to run the systemctl enable line.

TheTee82, 2021-02-05:
@tatokis 

>When the daemon is disabled, the gui should prompt you to run the systemctl enable line and show no connected devices.

* disabled the service and rebooted:
[image]

* ran `usb-devices` and checked the block you described:
```
T:  Bus=01 Lev=01 Prnt=01 Port=05 Cnt=03 Dev#=  3 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1b1c ProdID=1b55 Rev=03.24
S:  Manufacturer=Corsair
S:  Product=CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming Keyboard
S:  SerialNumber=some_hex_sn_guess_you_dont_need_it
C:  #Ifs= 2 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbhid
I:  If#=0x1 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
```

Note:
* this issue appeared only a month or so ago (sorry, didn't take down the exact date) the keyboard ran fine for about 2 months with ckb-next installed fine
* the issue appears to be completely random (at least from what I can tell) on boot: sometimes working, other times not
* on boot (with issue):
  * pre-grub: keyboard is using default settings
  * pre-xserver: keyboard goes dark for a second or so (guess that's were the driver initializes the kb)
  * post-xserver: keyboard goes back to default settings for a fraction of a second
    * then loads my custom default profile colors
    * does not respond to any input at all

Pillgar, 2021-02-07:
I think it's an issue with the keyboard, and not necessarily CKB-Next.  I've uninstalled the software, though it doesn't happen as frequently, the computer does still boot up without keyboard functionality (unplugging and connecting it again fixes it, or a reboot).  

It even happens in the grub menu, or trying to enter the bios (before any boot loader is detected).  This happened on two separate machines, and two separate keyboards (k70 mk2).

Also something weird, when S4/S5 states are disabled in bios, if I unplug the keyboard, the leds turn on after a few seconds.  The keyboard is receiving power through the USB pass-through connector??!!??

This could correlate with CKB-Next detecting two of the same device when the malfunction happens.

You could try to unplug the pass-through and see if that helps, but I'm not sure if it's intended to supplement 5volt power to the keyboard???

TheTee82, 2021-02-07:
@Pillgar 

>You could try to unplug the pass-through and see if that helps

Sure. 

However I'm fairly certain the Keyboard itself is fine since when the issue appears, selecting grub entries works fine (I've noticed since I often speed up the boot process in grub on my dual boot machine that way).

For now, I'm testing a (dirty) workaround:
* add following lines in `/etc/sudoers.d/myuser`:
```
myuser ALL=(root) NOPASSWD: /usr/bin/systemctl start ckb-next-daemon.service
myuser ALL=(root) NOPASSWD: /usr/bin/systemctl kill ckb-next-daemon.service
```
* create a script with following content:
```bash
#!/bin/sh

sudo /usr/bin/systemctl kill ckb-next-daemon.service
sleep 2  #wait a bit to give the systemd-unit some chance to close properly
sudo /usr/bin/systemctl start ckb-next-daemon.service

```
* put the script as executable in autostart (for me it's KDE so it shall be placed in `~/.config/autostart-scripts/`)
* I'm basically killing and restarting the daemon before the GUI application gets a chance to complain that the service is not running
  * I tried to change the autostart-command of `ckb-next` itself but it seems to self-correct as soon as it starts

Sure, this is a very ugly workaround but for now, this seems to do the trick (will edit my post if it does not since the issue appears to be random and I might have been just lucky for now).

TheTee82, 2021-02-07:
@tatokis 

>there's some application or another driver that's refusing to let go of the keyboard

I've checked systemd services but couldn't find any hint on what might be hogging the keyboard:

<details><summary>systemd units (system level)</summary>
<p>

```
sudo systemctl list-units --type=service      
  UNIT                               LOAD   ACTIVE SUB     DESCRIPTION                                                                  
  avahi-daemon.service               loaded active running Avahi mDNS/DNS-SD Stack                                                      
  bluetooth.service                  loaded active running Bluetooth service                                                            
  ckb-next-daemon.service            loaded active running Corsair Keyboards and Mice Daemon                                            
  cronie.service                     loaded active running Periodic Command Scheduler                                                   
  dbus.service                       loaded active running D-Bus System Message Bus                                                     
  kmod-static-nodes.service          loaded active exited  Create list of static device nodes for the current kernel                    
  lvm2-lvmetad.service               loaded active running LVM2 metadata daemon                                                         
  lvm2-monitor.service               loaded active exited  Monitoring of LVM2 mirrors, snapshots etc. using dmeventd or progress polling
  lvm2-pvscan@259:7.service          loaded active exited  LVM2 PV scan on device 259:7                                                 
  ModemManager.service               loaded active running Modem Manager                                                                
  NetworkManager-wait-online.service loaded active exited  Network Manager Wait Online                                                  
  NetworkManager.service             loaded active running Network Manager                                                              
  polkit.service                     loaded active running Authorization Manager                                                        
  preload.service                    loaded active running Adaptive readahead daemon                                                    
  rtkit-daemon.service               loaded active running RealtimeKit Scheduling Policy Service                                        
  sddm.service                       loaded active running Simple Desktop Display Manager                                               
  systemd-binfmt.service             loaded active exited  Set Up Additional Binary Formats                                             
  systemd-journal-flush.service      loaded active exited  Flush Journal to Persistent Storage                                          
  systemd-journald.service           loaded active running Journal Service                                                              
  systemd-logind.service             loaded active running User Login Management                                                        
  systemd-machined.service           loaded active running Virtual Machine and Container Registration Service                           
  systemd-modules-load.service       loaded active exited  Load Kernel Modules                                                          
  systemd-random-seed.service        loaded active exited  Load/Save Random Seed                                                        
  systemd-remount-fs.service         loaded active exited  Remount Root and Kernel File Systems                                         
  systemd-resolved.service           loaded active running Network Name Resolution                                                      
  systemd-sysctl.service             loaded active exited  Apply Kernel Variables                                                       
  systemd-timesyncd.service          loaded active running Network Time Synchronization                                                 
  systemd-tmpfiles-setup-dev.service loaded active exited  Create Static Device Nodes in /dev                                           
  systemd-tmpfiles-setup.service     loaded active exited  Create Volatile Files and Directories                                        
  systemd-udev-trigger.service       loaded active exited  Coldplug All udev Devices                                                    
  systemd-udevd.service              loaded active running Rule-based Manager for Device Events and Files                               
  systemd-update-utmp.service        loaded active exited  Update UTMP about System Boot/Shutdown                                       
  systemd-user-sessions.service      loaded active exited  Permit User Sessions                                                         
  tlp.service                        loaded active exited  TLP system startup/shutdown                                                  
  udisks2.service                    loaded active running Disk Manager                                                                 
  upower.service                     loaded active running Daemon for power management                                                  
[... cut ...]
  plasma-kglobalaccel.service loaded active running KDE Global Shortcuts Server
  pulseaudio.service          loaded active running Sound Service              
  ssh-agent.service           loaded active running SSH key agent   
```

</p>
</details>

I also checked autorun applications/scripts but other than default Nextcloud, yakuake, remmina and ckb-next application there's nothing on user level deployed. And they don't look like the type of application that might interfere with the ckb-next-daemon.

If you need any other log/info, just let me know.

tatokis, 2021-03-09:
I am still unable to reproduce this bug on different machines I tried.
As always, make sure you try the latest code from master to see if it helps. (The ckb-next-git package on AUR.)

As for logs, try `sudo journalctl -b 0 --unit=ckb-next-daemon --no-pager`

@TheTee82 Late, but I don't see anything that stands out there sadly...

TheTee82, 2021-03-09:
>Btw @TheTee82 Are you still running your workaround? Or did you find a way to resolve this?

kind of ...
the whole
```
sudo /usr/bin/systemctl kill ckb-next-daemon.service
sleep 2  #wait a bit to give the systemd-unit some chance to close properly
```
didn't work properly since when the issue happens, the daemon takes really long to properly stop (didn't take the time but I think at least a minute or so).

However what is working pretty well since a couple of weeks now:
* disable the service
```
systemctl disable ckb-next-daemon
```
* add a sudoers rule for your user to start the service
```
myuser ALL=(root) NOPASSWD: /usr/bin/systemctl start ckb-next-daemon.service
```
* create a script for starting the service
```
#!/bin/sh

sudo /usr/bin/systemctl restart ckb-next-daemon.service

exit 0
```
* `chmod +x` the script
* put it in autostart of your desktop environment (eg. KDE)

This starts the service before the GUI application is able to check whether the service is running or not.
Haven't had any issue with it since about 4 weeks now - before that it was every other day or so.

---

However, I'm curious @LtDrunk :
did your previously have any other keyboard led control software?
For me:
* I owned a Roccat Vulcan 120 Aimo and used this software for controlling the LEDs: https://github.com/duncanthrax/roccat-vulcan
* I also tested a Logitech G915 and tried https://aur.archlinux.org/packages/keyleds/ ... which didn't work out at all

TheTee82, 2021-03-09:
@tatokis 

>Late, but I don't see anything that stands out there sadly.

Yeah, me neither and might be something else I set up some time ago which is now interfering with the ckb-next-daemon but don't know what.
Anyway, as pointed out above, I'm fairly satisfied with the (ugly but seemingly effective) workaround of having the daemon in disabled state and starting it via user autostart.

I'll report back when I have the chance (and time) to dual-setup a clean environment on the same hardware to test. Cheers.

LtDrunk, 2021-03-09:
@tatokis 
> As for logs, try `sudo journalctl -b 0 --unit=ckb-next-daemon --no-pager`

`-- Journal begins at Mon 2020-07-20 16:34:01 CEST, ends at Tue 2021-03-09 21:26:47 CET. --
mar 09 19:07:01 linux-gaming systemd[1]: Started Corsair Keyboards and Mice Daemon.
mar 09 19:07:01 linux-gaming ckb-next-daemon[434]: ckb-next: Corsair RGB driver 0.4.3
mar 09 19:07:01 linux-gaming ckb-next-daemon[434]: [I] Root controller ready at /dev/input/ckb0
mar 09 19:07:01 linux-gaming ckb-next-daemon[434]: [I] Connecting CORSAIR K70 RGB MK.2 Mechanical Gaming Keyboard at /dev/input/ckb1
mar 09 19:07:01 linux-gaming ckb-next-daemon[434]: [I] Starting input thread for /dev/input/ckb1
mar 09 19:07:01 linux-gaming ckb-next-daemon[434]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 24 03 00 03 1c 1b 49 1b 01 01 00 01 c0 ff 40 01 03 00 00 00 02 40 02 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
mar 09 19:07:01 linux-gaming ckb-next-daemon[434]: [I] Setup finished for /dev/input/ckb1
mar 09 19:07:03 linux-gaming ckb-next-daemon[434]: [I] Connecting CORSAIR K70 RGB MK.2 Mechanical Gaming Keyboard at /dev/input/ckb2
mar 09 19:07:03 linux-gaming ckb-next-daemon[434]: [I] Starting input thread for /dev/input/ckb2
mar 09 19:07:03 linux-gaming ckb-next-daemon[434]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 24 03 00 03 1c 1b 49 1b 01 01 00 01 c0 ff 40 01 03 00 00 00 02 40 02 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
mar 09 19:07:03 linux-gaming ckb-next-daemon[434]: [I] Setup finished for /dev/input/ckb2
mar 09 19:07:13 linux-gaming ckb-next-daemon[434]: [E] os_sendindicators (usb_linux.c:280): Device or resource busy
mar 09 19:07:13 linux-gaming ckb-next-daemon[434]: [I] Attempting reset...
mar 09 19:07:13 linux-gaming ckb-next-daemon[434]: [E] os_sendindicators (usb_linux.c:280): Device or resource busy
mar 09 19:07:13 linux-gaming ckb-next-daemon[434]: [I] Attempting reset...
mar 09 19:07:14 linux-gaming ckb-next-daemon[434]: [I] Reset success
mar 09 19:07:14 linux-gaming ckb-next-daemon[434]: [E] usbclaim (usb_linux.c:522): Failed to claim interface 0: Device or resource busy
mar 09 19:07:14 linux-gaming ckb-next-daemon[434]: [E] os_resetusb (via usb.c:614): resetusb failed: Device or resource busy
mar 09 19:07:14 linux-gaming ckb-next-daemon[434]: [E] usb_tryreset (usb.c:622): Reset failed. Disconnecting.
mar 09 19:07:18 linux-gaming ckb-next-daemon[434]: [E] os_usbsend (via device_keyboard.c:40): Device or resource busy, res = 0xffffffff
mar 09 19:07:18 linux-gaming ckb-next-daemon[434]: [I] Attempting reset...
mar 09 19:07:19 linux-gaming ckb-next-daemon[434]: [E] os_usbsend (via device_keyboard.c:40): Device or resource busy, res = 0xffffffff
mar 09 19:07:19 linux-gaming ckb-next-daemon[434]: [I] Attempting reset...
mar 09 19:07:19 linux-gaming ckb-next-daemon[434]: [E] usbclaim (usb_linux.c:522): Failed to claim interface 1: Device or resource busy
mar 09 19:07:19 linux-gaming ckb-next-daemon[434]: [E] os_resetusb (via usb.c:614): resetusb failed: Device or resource busy
mar 09 19:07:19 linux-gaming ckb-next-daemon[434]: [E] usb_tryreset (usb.c:622): Reset failed. Disconnecting.
mar 09 19:07:19 linux-gaming ckb-next-daemon[434]: [I] Disconnecting /dev/input/ckb2
mar 09 19:07:19 linux-gaming ckb-next-daemon[434]: [I] Caught internal signal SIGUSR2
mar 09 19:07:19 linux-gaming ckb-next-daemon[434]: [I] Stopping indicator thread for ckb2 (4)
mar 09 19:07:20 linux-gaming ckb-next-daemon[434]: [I] Reset success
mar 09 19:07:20 linux-gaming ckb-next-daemon[434]: [I] Removed device path /dev/input/ckb2
mar 09 19:07:20 linux-gaming ckb-next-daemon[434]: [E] os_sendindicators (usb_linux.c:280): Device or resource busy
mar 09 19:07:20 linux-gaming ckb-next-daemon[434]: [I] Attempting reset...
mar 09 19:07:22 linux-gaming ckb-next-daemon[434]: [I] Reset success
mar 09 19:08:53 linux-gaming ckb-next-daemon[434]: [I] Stopping input thread for /dev/input/ckb1
mar 09 19:08:53 linux-gaming ckb-next-daemon[434]: [I] Stopping input thread for /dev/input/ckb2
mar 09 19:08:53 linux-gaming ckb-next-daemon[434]: [I] Disconnecting /dev/input/ckb1
mar 09 19:08:53 linux-gaming ckb-next-daemon[434]: [I] Caught internal signal SIGUSR2
mar 09 19:08:53 linux-gaming ckb-next-daemon[434]: [I] Stopping indicator thread for ckb1 (4)
mar 09 19:08:53 linux-gaming ckb-next-daemon[434]: [I] Removed device path /dev/input/ckb1
mar 09 19:09:08 linux-gaming ckb-next-daemon[434]: [I] Connecting CORSAIR K70 RGB MK.2 Mechanical Gaming Keyboard at /dev/input/ckb1
mar 09 19:09:08 linux-gaming ckb-next-daemon[434]: [I] Starting input thread for /dev/input/ckb1
mar 09 19:09:08 linux-gaming ckb-next-daemon[434]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 24 03 00 03 1c 1b 49 1b 01 01 00 01 c0 ff 40 01 03 00 00 00 02 00 02 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
mar 09 19:09:08 linux-gaming ckb-next-daemon[434]: [I] Setup finished for /dev/input/ckb1
`

@TheTee82 
> However, I'm curious @LtDrunk :
did your previously have any other keyboard led control software?

Nope, I did however run Manjaro for a couple of months without ckb, just having the standard LED-setup on the keyboard (I guess the one loaded from the keyboard memory).
So IF something is in conflict, the only possible thing would be the Linux/Manjaro-standard keyboard software/drivers if that was your thought

tatokis, 2021-03-09:
Can someone add a `#define DEBUG` near the top of usb_linux.c like so
```patch
diff --git a/src/daemon/usb_linux.c b/src/daemon/usb_linux.c
index dfa41cb..cad9b20 100644
--- a/src/daemon/usb_linux.c
+++ b/src/daemon/usb_linux.c
@@ -7,6 +7,8 @@
 #ifdef OS_LINUX
 #include <time.h>
 
+#define DEBUG
+
 // usb.c
 extern _Atomic int reset_stop;
 
```
and then recompile and send the logs when the daemon detects the device twice?

TheTee82, 2021-03-10:
@tatokis 

Here you go:
[image]

  </p>
  </details>
  
  
<details><summary>journalctl ckb-next-daemon</summary>
<p>

```
sudo journalctl -b 0 --unit=ckb-next-daemon --no-pager
-- Journal begins at Tue 2021-03-09 20:40:55 CET, ends at Wed 2021-03-10 07:19:12 CET. --
Mar 10 07:13:28 my-hostname systemd[1]: Started Corsair Keyboards and Mice Daemon.
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: ckb-next: Corsair RGB driver 0.4.3-7-ge48e6e6
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Root controller ready at /dev/input/ckb0
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] >>>vendor = 0x1b1c, product = 0x1b55, path = /dev/bus/usb/001/004, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Connecting CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming at /dev/input/ckb1
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Claiming interfaces. name=CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming Keyboard, firmware=0324, ep_str= 2
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] claiming 2 endpoints
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Starting input thread for /dev/input/ckb1
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Endpoint path /ep_81 has wMaxPacketSize 64
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 24 03 03 03 1c 1b 55 1b 01 01 00 01 c0 ff 40 00 03 00 00 00 02 40 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Setup finished for /dev/input/ckb1
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] >>>vendor = 0x1b1c, product = 0x1b55, path = /dev/bus/usb/001/004, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Connecting CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming at /dev/input/ckb2
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Claiming interfaces. name=CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming Keyboard, firmware=0324, ep_str= 2
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] claiming 2 endpoints
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Starting input thread for /dev/input/ckb2
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Endpoint path /ep_81 has wMaxPacketSize 64
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 24 03 03 03 1c 1b 55 1b 01 01 00 01 c0 ff 40 00 03 00 00 00 02 40 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Mar 10 07:13:28 my-hostname ckb-next-daemon[801]: [I] Setup finished for /dev/input/ckb2
Mar 10 07:13:31 my-hostname ckb-next-daemon[801]: [E] os_sendindicators (usb_linux.c:282): Device or resource busy
Mar 10 07:13:31 my-hostname ckb-next-daemon[801]: [I] Attempting reset...
Mar 10 07:13:32 my-hostname ckb-next-daemon[801]: [I] claiming 2 endpoints
Mar 10 07:13:32 my-hostname ckb-next-daemon[801]: [E] os_sendindicators (usb_linux.c:282): Device or resource busy
Mar 10 07:13:32 my-hostname ckb-next-daemon[801]: [I] Attempting reset...
Mar 10 07:13:32 my-hostname ckb-next-daemon[801]: [I] Reset success
Mar 10 07:13:32 my-hostname ckb-next-daemon[801]: [I] claiming 2 endpoints
Mar 10 07:13:32 my-hostname ckb-next-daemon[801]: [E] os_sendindicators (usb_linux.c:282): Cannot send after transport endpoint shutdown
Mar 10 07:13:32 my-hostname ckb-next-daemon[801]: [I] Attempting reset...
Mar 10 07:13:33 my-hostname ckb-next-daemon[801]: [I] claiming 2 endpoints
Mar 10 07:13:33 my-hostname ckb-next-daemon[801]: [I] Reset success
Mar 10 07:13:33 my-hostname ckb-next-daemon[801]: [I] Reset success
Mar 10 07:13:33 my-hostname ckb-next-daemon[801]: [E] os_sendindicators (usb_linux.c:282): Device or resource busy
Mar 10 07:13:33 my-hostname ckb-next-daemon[801]: [I] Attempting reset...
Mar 10 07:13:34 my-hostname ckb-next-daemon[801]: [I] claiming 2 endpoints
Mar 10 07:13:34 my-hostname ckb-next-daemon[801]: [E] os_usbsend (via led_keyboard.c:163): Device or resource busy, res = 0xffffffff
Mar 10 07:13:34 my-hostname ckb-next-daemon[801]: [I] Attempting reset...
Mar 10 07:13:34 my-hostname ckb-next-daemon[801]: [I] claiming 2 endpoints
Mar 10 07:13:34 my-hostname ckb-next-daemon[801]: [E] usbclaim (usb_linux.c:524): Failed to claim interface 1: Device or resource busy
Mar 10 07:13:34 my-hostname ckb-next-daemon[801]: [E] os_resetusb (via usb.c:614): resetusb failed: Device or resource busy
Mar 10 07:13:34 my-hostname ckb-next-daemon[801]: [E] usb_tryreset (usb.c:622): Reset failed. Disconnecting.
Mar 10 07:13:34 my-hostname ckb-next-daemon[801]: [E] os_usbsend (via device_keyboard.c:40): Device or resource busy, res = 0xffffffff
Mar 10 07:13:34 my-hostname ckb-next-daemon[801]: [I] Attempting reset...
Mar 10 07:13:35 my-hostname ckb-next-daemon[801]: [I] claiming 2 endpoints
Mar 10 07:13:35 my-hostname ckb-next-daemon[801]: [E] os_usbsend (via device_keyboard.c:40): Device or resource busy, res = 0xffffffff
Mar 10 07:13:35 my-hostname ckb-next-daemon[801]: [I] claiming 2 endpoints
Mar 10 07:13:35 my-hostname ckb-next-daemon[801]: [E] os_usbsend (via device_keyboard.c:40): Device or resource busy, res = 0xffffffff
Mar 10 07:13:36 my-hostname ckb-next-daemon[801]: [I] claiming 2 endpoints
[... cut ...]
#include "devnode.h"
#include "input.h"
#include "notify.h"
#include "usb.h"

#ifdef OS_LINUX
#include <time.h>

#define DEBUG

// usb.c
extern _Atomic int reset_stop;

/// \details
...
```
* recompiled and built using `./quickinstall`
* double-checked that service is enabled (which is was) `sudo systemctl enable ckb-next-daemon`
* restarted
  * the issue happened straight away
* took screenshot
* unplugged/replugged keyboard to get logs:
```
sudo journalctl -b 0 --unit=ckb-next-daemon --no-pager
```

tatokis, 2021-03-15:
Can you try once more the latest code from master?
This doesn't fix the root cause, but it fixes the code that's supposed to disallow duplicate devices.

https://github.com/ckb-next/ckb-next/commit/1af58f684b174e5c5da95eab41e90e1e59b361c2

It appears that for some reason the daemon gets an event that the device was connected, but never disconnected, which is quite odd.

TheTee82, 2021-03-16:
@tatokis 

>Can you try once more the latest code from master?

Awesome! :+1: 

This seems to have done the trick.
I disabled my modifications/scripts and installed latest master `0.4.3-27-g1af58f6` and rebooted 6 times (from power-off to standby and several reboots). And for now, the issue didn't happen a single time.

Here's the current systemd journal with `#define DEBUG` enabled:
```
-- Journal begins at Mon 2021-03-15 21:06:31 CET, ends at Tue 2021-03-16 07:53:10 CET. --
Mar 16 07:52:53 my-hostname systemd[1]: Started Corsair Keyboards and Mice Daemon.
Mar 16 07:52:53 my-hostname ckb-next-daemon[809]: ckb-next: Corsair RGB driver 0.4.3-27-g1af58f6
Mar 16 07:52:53 my-hostname ckb-next-daemon[809]: [I] Root controller ready at /dev/input/ckb0
Mar 16 07:52:53 my-hostname ckb-next-daemon[809]: [I] >>>vendor = 0x1b1c, product = 0x1b55, path = /dev/bus/usb/001/004, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6
Mar 16 07:52:54 my-hostname ckb-next-daemon[809]: [I] Connecting CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming at /dev/input/ckb1
Mar 16 07:52:54 my-hostname ckb-next-daemon[809]: [I] Claiming interfaces. name=CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming Keyboard, firmware=0324, ep_str= 2
Mar 16 07:52:54 my-hostname ckb-next-daemon[809]: [I] claiming 2 endpoints
Mar 16 07:52:54 my-hostname ckb-next-daemon[809]: [I] Starting input thread for /dev/input/ckb1
Mar 16 07:52:54 my-hostname ckb-next-daemon[809]: [I] Endpoint path /ep_81 has wMaxPacketSize 64
Mar 16 07:52:54 my-hostname ckb-next-daemon[809]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 24 03 03 03 1c 1b 55 1b 01 01 00 01 c0 ff 40 00 03 00 00 00 02 40 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Mar 16 07:52:54 my-hostname ckb-next-daemon[809]: [I] Setup finished for /dev/input/ckb1
Mar 16 07:52:54 my-hostname ckb-next-daemon[809]: [I] >>>vendor = 0x1b1c, product = 0x1b55, path = /dev/bus/usb/001/004, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6
Mar 16 07:52:54 my-hostname ckb-next-daemon[809]: [W] usbadd (usb_linux.c:682): Attempted to add device with duplicate syspath /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6 (path /dev/bus/usb/001/004)
```

I'll keep you posted in case the issue pops up again and of course will try to help debugging in case you want to go after the root cause.
I really appreciate what you have done (even dropped my old mouse in favor of a corsair mouse since this application is really nice!)

Cheers.

LtDrunk, 2021-03-16:
> Can you try once more the latest code from master?

Can confirm this fixed my issue aswell!
Did an uninstall the AUR git-package.
Cloned the latest git and did a manual build, and this version is working.

The ckb-next-daemon is enabled by default and ckb-next is up-and-running. No need to unplug & plug in the keyboard anymore
 :D
 
 Also, here is my output from the `sudo journalctl -b 0 --unit=ckb-next-daemon --no-pager` command on the working version if it will help in any way
 
`-- Journal begins at Mon 2020-07-20 16:34:01 CEST, ends at Tue 2021-03-16 09:51:39 CET. --
mar 16 09:46:49 linux-gaming systemd[1]: Started Corsair Keyboards and Mice Daemon.
mar 16 09:46:49 linux-gaming ckb-next-daemon[450]: ckb-next: Corsair RGB driver 0.4.3-27-g1af58f6
mar 16 09:46:49 linux-gaming ckb-next-daemon[450]: [I] Root controller ready at /dev/input/ckb0
mar 16 09:46:49 linux-gaming ckb-next-daemon[450]: [I] Connecting CORSAIR K70 RGB MK.2 Mechanical Gaming Keyboard at /dev/input/ckb1
mar 16 09:46:50 linux-gaming ckb-next-daemon[450]: [I] Starting input thread for /dev/input/ckb1
mar 16 09:46:50 linux-gaming ckb-next-daemon[450]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 24 03 00 03 1c 1b 49 1b 01 01 00 01 c0 ff 40 01 03 00 00 00 02 00 02 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
mar 16 09:46:50 linux-gaming ckb-next-daemon[450]: [I] Setup finished for /dev/input/ckb1
mar 16 09:46:50 linux-gaming ckb-next-daemon[450]: [W] usbadd (usb_linux.c:679): Attempted to add device with duplicate syspath /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6 (path /dev/bus/usb/001/003)
`

tatokis, 2021-03-18:
I made one more change. Please pull again and test the code to make sure that it still works.
Additionally, if you don't mind, try applying this on top and send the log back.
```patch
diff --git a/src/daemon/usb_linux.c b/src/daemon/usb_linux.c
index cc3c215..a12a7eb 100644
--- a/src/daemon/usb_linux.c
+++ b/src/daemon/usb_linux.c
@@ -903,6 +903,7 @@ int usbmain(){
                 continue;
             }
             // Add/remove device
+            ckb_info("Device action %s", action);
             if(!strcmp(action, "add")){
                 int res = usb_add_device(dev);
                 if(res == 0)

```

TheTee82, 2021-03-18:
@tatokis 

done.
Tested a couple of reboots and power-offs and still seems to work fine.
Here's the log you requested:
```
-- Journal begins at Thu 2021-03-18 07:45:58 CET, ends at Thu 2021-03-18 09:34:08 CET. --
Mar 18 09:33:47 my-hostname systemd[1]: Started Corsair Keyboards and Mice Daemon.
Mar 18 09:33:47 my-hostname ckb-next-daemon[756]: ckb-next: Corsair RGB driver 0.4.3-28-geea21d4
Mar 18 09:33:47 my-hostname ckb-next-daemon[756]: [I] Root controller ready at /dev/input/ckb0
Mar 18 09:33:47 my-hostname ckb-next-daemon[756]: [I] >>>vendor = 0x1b1c, product = 0x1b55, path = /dev/bus/usb/001/003, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Connecting CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming at /dev/input/ckb1
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Claiming interfaces. name=CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming Keyboard, firmware=0324, ep_str= 2
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] claiming 2 endpoints
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Starting input thread for /dev/input/ckb1
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Endpoint path /ep_81 has wMaxPacketSize 64
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 24 03 03 03 1c 1b 55 1b 01 01 00 01 c0 ff 40 00 03 00 00 00 02 40 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Setup finished for /dev/input/ckb1
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] >>>vendor = 0x1b1c, product = 0x1b55, path = /dev/bus/usb/001/003, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [W] usbadd (usb_linux.c:682): Attempted to add device with duplicate syspath /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6 (path /dev/bus/usb/001/003)
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action change
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action change
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:48 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action change
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action add
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action bind
Mar 18 09:33:49 my-hostname ckb-next-daemon[756]: [I] Device action bind
```

TheTee82, 2021-04-23:
Just wanted to give some feedback since I was testing the changes using the manually built version **and** the latest package provided in AUR (`aur/ckb-next 0.4.4-1`).

The issue seems to be solved, at least for my device and everything works perfectly fine after the fix was pushed ( https://github.com/ckb-next/ckb-next/commit/1af58f684b174e5c5da95eab41e90e1e59b361c2 ).

Not sure what your usual github-issue-workflow is but as far as I'm concerned this issue can be closed.

closed 2022-05-29 by Ravenslofty

CodexHere, 2023-06-09:
Just found this issue, and it's been bothering me for MONTHS now.

Running `0.5.0-1` and it's still having this issue. Having to disable the service and start on autologin to resolve the issue.