ckb-next/ckb-next #669: Add support for dark core pro please

tournade, 2020-11-29:
hello I have a new mouse the dark core rgb pro and cbk does not support it at the moment

is it possible to add it in your software

thank you

ps .: I send you the ckb-next-dev-dectect

thanks again
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/5612825/ckb-next-dev-detect-report.gz)

Electro707, 2020-12-05:
Yup this is another bragi device, same issue as #575

Mjolinir, 2021-01-26:
I would also like to see support for this device. I plug it in via USB, not wirelessly. I attached ckb-next-dev-dectect report.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/5875016/ckb-next-dev-detect-report.gz)

gravityfargo, 2024-05-02:
If you open a new report, please run the report with the mouse plugged in and not using the wireless dongle. 

This issue has been marked as stale due to age and/or lack of activity. 

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-02 by gravityfargo