ckb-next/ckb-next #671: Keyboard not working on Wayland

matywastaken, 2020-12-03:
## General information
Since the last update (includes systemd 247.1-1), starting the ckb daemon prevents the keyboard working on Wayland (Sway)
Everything works in the TTY, executing sway creates this issue.
The bug occurs also if the daemon is enabled after the session is started.

### Source
ckb-next package from the AUR

### Logs & crash reports

[ckb-next-report-1606954067.gz](https://github.com/ckb-next/ckb-next/files/5632818/ckb-next-report-1606954067.gz)

tatokis, 2020-12-03:
The daemon only talks to the kernel via uinput and has nothing to do with wayland. It's probably not our bug.
That said, I don't see anything about the daemon in your report. This sounds like either a libinput or a sway bug.
Check whatever the equivalent of `Xorg.0.log` is on sway.

Additionally, if you updated the kernel, you should reboot if you haven't already.

kuon, 2020-12-04:
I had the same issue after upgrade, I installed `ckb-next-git` under arch and the issue was gone.

I am not sure if is the git version or the the recompilation (some lib might have changed).

tatokis, 2020-12-04:
I just tested this and I can not reproduce this with the Xorg server and libinput in a VM.

In the git package, the daemon presents the devices differently. Previously they used to have every single available button, key and axis enabled, but now keyboards only have keys and mice only buttons and axes.
I suspect this is related, but nonetheless, this is a bug elsewhere.

If you can figure out which package caused the regression, and you're interested in opening a bug on their tracker, you can refer to these commits as a possible reason the `ckb-next-git` package works compared to the 0.4.2 release.

https://github.com/ckb-next/ckb-next/commit/0ae46be1dd4151b7b5dc80a0271d5a86b8826af1
https://github.com/ckb-next/ckb-next/commit/afd0e19f1d198b3807d137be40cb845a28ccd2d5

kuon, 2020-12-04:
Well, it might be a specific wayland issue. I am not familiar with how everything is stitched together (wayland, xwayland, libinput...).

I looked at yesterday's upgrade that showed the issue, I updated about 600 packages, and in this list there is nearly all xorg packages (xorg-server-xwayland (1.20.9.r21.g5c400cae1-2 -> 1.20.10-1) and friends). So I guess it might be the issue.

But well, you commits does patch the issue, so I guess it's ok, if @matywastaken can confirm using the git version fixes the issue too I think we can close this and reference to it if we find upstream culprit.

matywastaken, 2020-12-04:
Installing the ckb-next-git package solved the problem, xwayland and other xorg packages were updated before the issue started happening, but i don't know how everything fits in this case.

This issue does not belong here then, if i find something else as to what is the cause i'll reference this, for now i'm not savvy enough to point anywhere.

Thanks for the help!

closed 2022-10-03 by tatokis