ckb-next/ckb-next #674: Start hidden

XReaper95, 2020-12-17:
## What is the program's current behaviour?
I have this program to start automatically at login. I know this is a very minor thing but I just can't find a way to make it start hidden (minimized), maybe I'm missing something, or isn't really possible, in which case I think it should be a nice feature. I know I can just minimize it every time, but this is triggering my OCD xD.

## What would you like it to do instead?
To start minimized.

## Any other notes & comments?
My system specs:
```
 OS: 5.4.80-2-MANJARO
 Desktop Environment: KDE
```

tatokis, 2020-12-17:
It should be fixed in the `ckb-next-git` package. Give that a try.

XReaper95, 2020-12-17:
It is indeed! Thank you @tatokis. I can close this now.

closed 2020-12-17 by XReaper95