ckb-next/ckb-next #676: Holding a button down should not trigger a macro multiple times

tguen, 2020-12-18:
## General information
I wanted to configure the dpi buttons on my Nightsword to send `ctrl+pgup` and `ctrl+pgdown` so I can use them to switch tabs. I ended up having to use macros (see #675). This works as expected most of the time (aside from lag when pressing multiple times in quick succession, does this warrant another issue?), but if I accidentally hold the button down for a fraction of a second, the macro is triggered multiple times and I skip over tabs.

### Source
AUR (tried both `ckb-next` and `ckb-next-git`)

## Logs & crash reports
[ckb-next-report-1608282325.gz](https://github.com/ckb-next/ckb-next/files/5714723/ckb-next-report-1608282325.gz)

## What is the program's current behaviour?
Clicking and holding a button with a macro assigned will trigger the macro multiple times

## What is the expected behaviour of the program?
The macro should only be triggered once, on button press or release.

JarnoKai, 2020-12-18:
This is possibly a new feature released just 9 hours ago in 0.4.3:

> New features:
> 
>     Under X11 only, lights can now automatically turn off after a user-set time
>     **Macros now loop when the key is held down**

tguen, 2020-12-22:
Interesting... I don't know why this would be desirable, but I suppose it would be nice as a setting. Not being able to disable it makes  my macros not work as desired. It would also be nice to be able to adjust how long the button must be held before it starts looping.

ZetaDaemon, 2021-03-01:
There seems to be a delay before the macro starts to loop, as it'll activate once, then a delay, then it'll quickly repeat the macro while the key or button is held down. This doesn't seem to be affected by repeat keys as turning it on or off and changing the delay doesn't effect this delay before looping. Is it possible that this delay before looping started could be changed

dgcampea, 2021-03-29:
It should be configurable behavior (for macros that act like turbo buttons).

> There seems to be a delay before the macro starts to loop, as it'll activate once, then a delay, then it'll quickly repeat the macro while the key or button is held down. This doesn't seem to be [...]

This is (intentional) due to https://github.com/ckb-next/ckb-next/commit/08082240e8798fd7df4b43778dd236c7fb3bdb25, checking out v0.4.3 and reverting it would solve it for this situation but a proper fix would be to have a local macro/keybind option that allows overriding this `firstloop` behavior.