ckb-next/ckb-next #684: Where to configurable idle time for light?

Shinigami92, 2020-12-22:
## General information

Since 0.4.3 my keyboard lightings turn off after some minutes
https://github.com/ckb-next/ckb-next/blob/master/CHANGELOG.md#v043-2020-12-18

```
New features:

-> Under X11 only, lights can now automatically turn off after a user-set time
```

Seems to be a feature and yeah I'm okay with it, but where can I configure it and bump up from minutes to e.g. hours? Or simply turn off this feature?

### Source
How and where did you get this program, e.g.: AUR

## What is the program's current behaviour?
Turn off lights after some minutes

## What is the expected behaviour of the program?
UI let me configure the feature

## Any other notes & comments?
Especially when I switch into my qemu Win10 VM for gaming, it turns off while I'm playing :eyes:  
So the tool don't receive any new key-inputs while playing in my VM (using left and right `ctrl` to switch between VM and arch linux)

tatokis, 2020-12-23:
While you do bring up a valid point, that we should also check for input from devices as well to determine that the system is idle, it can be configured in the settings tab...

[image]

Shinigami92, 2020-12-23:
Thank you for the screenshot
I needed a full restart of my system
And the window was a bit smaller on my screen, so I needed to resize it and didn't see the scrollbar directly

Now it's perfectly working, thx

closed 2020-12-23 by Shinigami92